mandulaj/PZEM-004T-v30 #1: Compatibility problem with Arduino Due

Terdious, 2019-05-25:
Hi,
I was able to test your library on an Arduino Mega. It works very well and thank you for your update. Unfortunately, only the Arduino Due is able to run my home automation programs and I have a compatibility problem when I wear it on this platform.

Thanks in advance,
cordially

mandulaj, 2019-10-01:
Hi there, sorry, I don't have an Arduino Due so I can't have a look at this. Maybe someone else can help?

Bruno999B, 2020-03-03:
Hello, I have the same problem, the problem is as follows:

arduino_due_x/variant.h:251:18: error: 'Serial' has a previous declaration as 'UARTClass Serial'
 extern UARTClass Serial;
invalid conversion from 'int' to 'HardwareSerial*' [-fpermissive]

you can see the problem by compiling with the IDE  configured on a due card.
Please take a look if you have a few minutes. I want to connect several Pzem on a Due, the mega is not powerful enough for the web interface...

Thank you!

mandulaj, 2020-04-21:
Has anyone found a solution, or work around? Is this issues still relevant?

Fox2829, 2020-11-05:
Hi,

Unfortunately the problem has not been resolved. 
Please help to implement compatibility with DUE.
It is very necessary!

Thank you!

mandulaj, 2020-11-05:
This issue might be related to #39 It looks like the workaround for using both hardware and software serial is not completely compatible with all boards. Further investigation is required thought.

seramonte, 2020-11-10:
Newbie, here. pzem017v1 library (for PZEM-017 DC meter) is identical to pzem004t30 except for specific data addresses. So this should apply. For pzem017v1  DUE compatibility I took inspiration from ModbusMaster library; it uses Stream instead of HardwareSerial; it separates initialization (outside of setup()) from begin() (inside setup). This works...except that oddly I get occasional random read errors (no problem when using ModbusMaster). Still investigating. This is what I did:

In PZEM017v1.cpp replace class initialization and add new begin() member:  
```
//extern HardwareSerial Serial;  // Commented out for DUE

/*
PZEM017v1::PZEM017v1(HardwareSerial* port, uint8_t addr) // Commented out for DUE
{
    port->begin(PZEM_BAUD_RATE, SERIAL_8N2);
    this->_serial = &port;
    this->_isSoft = false;
    init(addr);
}
*/

PZEM017v1::PZEM017v1(void)
{
}

void PZEM017v1::begin(uint8_t slaveAddr, Stream &serial)
{
    _serial = &serial;
    init(slaveAddr);
}
```
In pzem017v1.h:
```
    //PZEM017v1(HardwareSerial* port, uint8_t addr=PZEM_DEFAULT_ADDR);
    PZEM017v1(void); // changed for DUE

    void begin(uint8_t slaveAddr, Stream &serial); //  added
```
In .ino:
```
//PZEM017v1 pzem(&Serial3, slaveAddr);
PZEM017v1 pzem;

setup() {
  Serial3.begin(9600, SERIAL_8N2);
  pzem.begin(slaveAddr, Serial3);

...
```
I think that's it.