mandulaj/PZEM-004T-v30 #3: Hardserial on bluepill doesn't work

aarif29, 2019-06-30:
Hi everyone!

I just got the pzem v3 module and found this library (I use hardserial examples, because like in pzem v2 to use bluepill, we can only use hardserial libraries). I use Bluepill as MCU, and use serial 1 on PA9 & PA10 pins, but Sketch can't be uploaded and error messages like this appear:

Arduino: 1.8.5 (Windows 10), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), STLink, 72Mhz (Normal), Smallest (default)"
C: \ Users \ Windows 10 \ Documents \ Arduino \ libraries \ PZEM-004T-v30-master \ PZEM004Tv30.cpp: 32: 23: error: conflicting declaration 'HardwareSerial Serial'
 extern Hardware Serial Serial;
                       ^
In file included from C: \ Users \ Windows 10 \ Documents \ Arduino \ hardware \ Arduino_STM32-master \ STM32F1 \ cores \ maple / wirish.h: 71: 0,
                 from C: \ Users \ Windows 10 \ Documents \ Arduino \ hardware \ Arduino_STM32-master \ STM32F1 \ cores \ maple / Arduino.h: 30,
                 from C: \ Users \ Windows 10 \ Documents \ Arduino \ libraries \ PZEM-004T-v30-master \ PZEM004Tv30.h: 19,
                 from C: \ Users \ Windows 10 \ Documents \ Arduino \ libraries \ PZEM-004T-v30-master \ PZEM004Tv30.cpp: 1:
C: \ Users \ Windows 10 \ Documents \ Arduino \ hardware \ Arduino_STM32-master \ STM32F1 \ cores \ maple / usb_serial.h: 96: 22: error: 'Serial' has a previous declaration as 'USBSerial Serial'
     extern USB Serial Serial;
                      ^
C: \ Users \ Windows 10 \ Documents \ Arduino \ libraries \ PZEM-004T-v30-master \ PZEM004Tv30.cpp: In constructor 'PZEM004Tv30 :: PZEM004Tv30 (uint8_t, uint8_t, uint8_t)':
C: \ Users \ Windows 10 \ Documents \ Arduino \ libraries \ PZEM-004T-v30-master \ PZEM004Tv30.cpp: 62: 5: error: 'Software Serial' was not declared in this scope
     Software Serial * port = new Software Serial (receivePin, transmitPin);
     ^
C: \ Users \ Windows 10 \ Documents \ Arduino \ libraries \ PZEM-004T-v30-master \ PZEM004Tv30.cpp: 62: 21: error: 'port' was not declared in this scope
     Software Serial * port = new Software Serial (receivePin, transmitPin);
                     ^
C: \ Users \ Windows 10 \ Documents \ Arduino \ libraries \ PZEM-004T-v30-master \ PZEM004Tv30.cpp: 62: 32: error: expected type-specifier before 'SoftwareSerial'
     Software Serial * port = new Software Serial (receivePin, transmitPin);
                                ^
C: \ Users \ Windows 10 \ Documents \ Arduino \ libraries \ PZEM-004T-v30-master \ PZEM004Tv30.cpp: 62: 32: error: expected ';' before 'Software Serial'
exit status 1
Error compiling for Generic STM32F103C series board.

how is the solution to solve this problem?
I'm not good at programming for now

by the way, thank you for making this library for the v3 pzem module, hopefully it will get better in the future
Thank you!

mandulaj, 2019-10-01:
I can confirm the library does not work on Viagra, but I ordered the STM32F103C and will have a look if I have time :smiley:

mandulaj, 2020-04-21:
@aarif29 Have you managed to get it working or found some work around?

aarif29, 2020-04-21:
sorry I have not found a solution yet, then I decided to buy an old module that's version 2, so my project was done quickly, lol
however, in the future I want to try again to redevelop my project, and I will try to use version 3 module
thank you for have worked hard with this amazing library

tomashulman, 2021-05-16:
Hi,
I had the same problem with pzem 004T v3.0 on STM32. After quick check I found out that after set baudrate
to serial port by library STM32 freezes. Baudrate has to be set at SETUP() {   } section.

I am using hardware Serial2 {PA2=Tx , PA3=Rx}

Dirty but working solution in short:

Edit library PZEM004Tv30.cpp and add disable 2 lines "//" so it looks like

Line 54:  //extern HardwareSerial Serial;
Line 103: //port->begin(PZEM_BAUD_RATE);

Arduino IDE add:

void setup() {
.
Serial2.begin(9600);
.
}

Note: Do not forget to power 230VAC measure side. IC on module is powered from this side.
Iv been confused when data connector has a +5V pin and module did not work until I connected 230VAC to AC measure terminals.

Tested on Chinese Bluepill and Blackpill STM32F103C8T6 and STM32 mini V2.0 module STM32F103RCT6 with Roger`s bootloaders.