mandulaj/PZEM-004T-v30 #4: PZEM-004T v3.0 Power and Energy meter" To save data to SD CARD

pye222, 2019-07-10:
"PZEM-004T v3.0 Power and Energy meter" is used very well.

But there is one thing that is not good.

It wants to store the data value in SD CARD.
  SD CARD stores data by SPI communication.

It works well if only one SD card is connected to Arduino.
Or "PZEM-004T v3.0 Power and Energy meter".

The problem is that if both are connected, it will not work properly.
Is there any way?

Alfonso-Lijo, 2019-08-19:
I have a same problem! Now ! tri to put a external power supply.

Alfonso-Lijo, 2019-08-19:
I just change the PZEM004T.h line 34 (#define PZEM_DEFAULT_ADDR 0x0F to 0x001 and work now!

Alfonso-Lijo, 2019-08-19:
This library needs tweaking!
I've managed to keep up communication when I add the RTC and LCD control libraries, but sending anything to the LCD just ends communication!

Alfonso-Lijo, 2019-08-24:
Success!
I was able to perform all the steps using Arduino Mega!
The reason is quite obvious! Arduino Uno has only one serial port and simulates other serials on digital pins!
But the arduino Mega has 5 independent serial ports!
But attention: I do not know if I made a mistake or is a failure in the library, but only managed from the HardSerial example!
Follow my code!
I use an RTC1307, an SD card reader and a 20x4 LCD on my Datalogger!

/*  **************************************************************************************************
   SD - Se comunica por SPI
   MOSI -> pino 51
   MISO -> pino 50
   SCK -> pino 52
   CS (SS) -> Pino 53

   Relógio RTC DS1307 - Se comunica por I2C no endereço 0x68
   SDA -> pino 20
   SCL -> pino 21

   LCD se comunica por IC2 no endereço 0x27
   SDA -> pino 20
   SCL -> pino 21

   Coleta de dados do Pzem 004
   Pinos de comunicação Pzem (módulo Max485 RS 232 para Modbus)
   Pino 14 TX fio verde
   Pino 15 RX fio branco

   Liga Backligth do LCD
   pino digital 3 aciona o backligth
*/
#include <LiquidCrystal_I2C.h>  // Biblioteca de configuração do LCD
LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Inicializa o display no endereco 0x27
#include <SdFat.h>      //Uso a biblioteca SdFat no lugar da original do arduino pois é mais otimizada.
SdFat SD;               //Linha necessária para compatibilizar o programa escrito para SD.h (biblioteca original)
const int CS_SD = 53;     //Pino do cartão SD
#include <DS1307RTC.h>    //Biblioteca para controle do RTC DS1307
#include <PZEM004Tv30.h>     //Biblioteca do monitor de energia Pzem
PZEM004Tv30 pzem(&Serial3); // TX(Verde),RX(Branco) conectados para RX,TX do PZEM

// Constantes variadas

const unsigned long intervaloLeitura = 2000;      //inervalo de leitura [milisegundos]
const unsigned long intervaloEscritaSD = 300;     //intervalo de escrita de dados no SD [milisegundos]
boolean temSD = 0;        // Variável para indicar presença de cartão SD
const int intervaloSincro = 300; //Intervalo de sincronismo do arduino com o RTC [Seg.]. Sincronismo do RTC
//----------------------------------------------------------------------------------------------------------------

void setup()
{
  lcd.begin (20, 4);    //definição do LCD 20 colunas e 4 linhas
  Serial.begin(115200);   //Inicia a comunicação em 9600 kbps

  //Inicializa cartão SD
  // verifica se o SD está presente e pode ser inicializado:
  if (!SD.begin(CS_SD)) {
    temSD = 0;  //Cartão SD não detectado
    Serial.print("Cartão SD não detectado");
    Serial.println();
  }
  else {
    temSD = 1;  //Cartão SD encontrado
    Serial.print("Cartão está presente");
    Serial.println();
  }
  //Inicializa o RTC e contagem de tempo
  setTime(RTC.get());
  setSyncProvider(RTC.get); //Estabelecemos o RTC como provedor do tempo
  setSyncInterval(intervaloSincro); // Atualizamos seguindo intervalo definido
}
[... 199 lines cut ...]
  }
  if (SD.exists(fileName)) { //Se existe o arquivo "fileName" o abrimos e adicionamos os dados.
    File meulog = SD.open(fileName, FILE_WRITE);
    if (meulog) {           //Se o arquivo se abre corretamente escrevemos nele ...
      meulog.println(data);
      meulog.close();
      return 0;
    }
    else {
      return 1;   //Não se conseguiu abrir o arquivo ...
    }
  }
  else {                      //se não existe o archivo "meulog.cvs" criamos um novo e gravamos os dados ...
    File meulog = SD.open(fileName, FILE_WRITE);
    if (meulog) {           //Se o arquivo é aberto corretamente gravamos os dados ...
      meulog.println(F("Data, Hora, Volt, Amp, Watt, kWh, Hz, FP"));
      meulog.println(data);
      meulog.close();
      return 0;
    }
    else {
      return 1;
    }
  }
}

Alfonso-Lijo, 2019-08-24:
Thank you so much mandulaj for the beautiful work!

pye222, 2019-09-08:
I'm glad that it works well with mega.
I did software serial communication with two Arduinos. And one side
PZEM-004T v3.0 was installed and data was transferred to another Arduino. Other Arduinos were used by connecting Sd cards, LCDs, and RTCs.

Alfonso-Lijo, 2019-09-18:
I am a beginner in Arduino!
I am trying to improve my code so that the data sent to the LCD does not interfere with the data recording!
As it stands today, LCD display time directly interferes with the amount of recorded data!
Could someone help me sort this out so that I can record more data on the SD card while keeping the display time of the data on the LCD?
I would be very grateful!

Alfonso-Lijo, 2019-09-30:
Hello!
It works and very well!
I made a three phase datalogger reading three Pzem in serial 1, 2 and 3.
Without any interference or failure I am logging the data every second!
Fantastic!
Thank you very much!

mandulaj, 2019-10-01:
@Alfonso-Lijo, thanks for using my library, I wasn't expecting many people to do so. I was having similar issues and honestly I have no idea how to make it work more elegantly on an UNO . If you have any suggestions, please feel free to open up a pull request!

RB3rg, 2020-02-27:
@mandulaj I would not waste time with software serial, we all know it is a workaround for old devices.
Most of all new devices (esp32, stm32) have more than one HW Serial.
In the real-life, no one will (or should) put in production a device for monitoring power using software serial. It ends up most of the people using SW Serial it's bench test.

mandulaj, 2020-04-21:
@pye222 Is this still something that needs work? Have you found a way to fix or some kind of work around?

Alfonso-Lijo, 2020-04-21:
It's working within my needs and I haven't made any more changes! Thanks!

mandulaj, 2020-04-21:
Ok, in that case I will close this issue.

closed 2020-04-21 by mandulaj

nmsneves, 2020-11-05:
Hi, I've successfuly used your library with arduino uno, but I tried to compile for MKR 1010 wifi, without success. Neither software or hardware serial. Is this device supported?