mandulaj/PZEM-004T-v30 #5: The question is a lot of targeting. What if I use more than one device?

kkwestt, 2019-07-18:
The question is a lot of targeting. 
What if I use more than one device?
protocol have multichannel addressing

pmoser56, 2019-08-05:
Since you can not identfiy the devices on a single interface, use multiple interfaces.

mandulaj, 2019-10-01:
@kkwestt, you can give each device an address and then create an instance for each device with that address. See the [ PZEMChangeAddress example](https://github.com/mandulaj/PZEM-004T-v30/tree/master/examples/PZEMChangeAddress) 

You can create an instance with a specific address as `PZEM004Tv30 pzem(11, 12, <address>);`

I hope this helps. If you have any more questions, don't hesitate to ask!

closed 2019-10-01 by mandulaj

luispopity, 2019-11-20:
I could put an example with 2 modules operating on the same port, thanks

mandulaj, 2019-11-20:
What you could do is use the 'change-address' program to give each device a specific address. Then in your actual program you can pass this address to the pzem instance as I have shown above. `PZEM004Tv30 pzem(11, 12, <address>);`

luispopity, 2019-11-21:
ok, I did it, but first put the address with the software changeadress, thanks