mandulaj/PZEM-004T-v30 #9: Library

pmoser56, 2019-08-05:
Thanks a lot for your library. It works really well.
What does not work is the usage of "NAN" in the SoftwareSerial example. I replaced it with "isnan" and then it works really well so far. I created a 3-phase circuit with an esp8266 with 3 SoftwareSerial interfaces and mqtt readout. Thanks a lot for your work.

mandulaj, 2019-10-01:
It looks like SoftwareSerial is not as hardware independent as I thought. I am considering rewriting it anyways. Thanks!

mandulaj, 2019-10-01:
Yeah I had a second look and now I see the problem. I will change this, thanks for pointing it out.

frthecrow, 2019-10-05:
Hellow
Have you try it with Mysensors.
I have a probleme with this code.
Could you help me ?

[Watmetre_Mysensors_2019-10-04.zip](https://github.com/mandulaj/PZEM-004T-v30/files/3693347/Watmetre_Mysensors_2019-10-04.zip)

thanks

mandulaj, 2019-10-05:
> Hellow
> Have you try it with Mysensors.
> I have a probleme with this code.
> Could you help me ?
> 
> [Watmetre_Mysensors_2019-10-04.zip](https://github.com/mandulaj/PZEM-004T-v30/files/3693347/Watmetre_Mysensors_2019-10-04.zip)
> 
> thanks

Hi, could you please specify more details regarding your problem? Are you not getting the right values? No values at all? Errors? Does it not compile? I can't compile your code as there are some files missing... You can paste the results from your console and then I can help you but like this, there are way too many variables that could be the cause of your issue...
cheers!

frthecrow, 2019-10-05:
Hi 

Picture for speeking.

Thansk

Provenance : Courrier pour Windows 10

De : Jakub Mandula
Envoyé le :samedi 5 octobre 2019 17:14
À : mandulaj/PZEM-004T-v30
Cc : frthecrow; Comment
Objet :Re: [mandulaj/PZEM-004T-v30] Library (#9)

Hellow
Have you try it with Mysensors.
I have a probleme with this code.
Could you help me ?
Watmetre_Mysensors_2019-10-04.zip
thanks
Hi, could you please specify more details regarding your problem? Are you not getting the right values? No values at all? Errors? Does it not compile? I can't compile your code as there are some files missing... You can paste the results from your console and then I can help you but like this, there are way too many variables that could be the cause of your issue...
cheers!

frthecrow, 2019-10-05:
[image]
[image]
[image]
[image]

frthecrow, 2019-10-06:
Hello
Ok With an other computer.

Excuse me
thanks

ruimiguelsa, 2019-10-24:
> Thanks a lot for your library. It works really well.
> What does not work is the usage of "NAN" in the SoftwareSerial example. I replaced it with "isnan" and then it works really well so far. I [...]

Hello, 
Can you post your code? Thanks in advance.

mandulaj, 2019-11-19:
I think now this issue is resolved in #14

closed 2019-11-19 by mandulaj