mandulaj/PZEM-004T-v30 #10: 'Serial3' was not declared in this scope

NepNutella, 2019-08-06:
Hi, I copied and pasted the code, but the error console is " 'Serial3' was not declared in this scope "..
i using:
Arduino uno, pzem-004T-v30

Thank you!

[image]

sergiocntr, 2019-08-06:
Serial3 is not available on Arduino Uno,you can use Serial instead. You can follow two routes,the first ,use hardware serial,second use software serial.  I prefer to leave hardware serial for debugging and software serial for pzem interfacing. Each two are easy to implement.

NepNutella, 2019-08-06:
Oh thanks for the reply. 
Do you have some links to explain this to me?
Thanx so much!

sergiocntr, 2019-08-06:
You are welcome! That so easy....don't be scared ! Personal thoughts and sorry to say that there are tons of tutorial ,instruction , diy about that and this is not the right place where to ask this question. I know that first steps are hard but you have to do by yourself,I hope you understand. Look on youtube, instructables...You'll find what you are looking for and much more else!!!

NepNutella, 2019-08-06:
I don't want a link with a code to copy and paste, I wanted a string to search on google!

sergiocntr, 2019-08-06:
"Software serial arduino uno" maybe? Have you tried? Is this so hard?

NepNutella, 2019-08-06:
I tried to search but it didn't specifically deal with the topic ... I'm sorry I'll try better

mandulaj, 2019-10-01:
I see you guys figured it out. Closing this

closed 2019-10-01 by mandulaj