mandulaj/PZEM-004T-v30 #12: Add calibration function to the library

devicter, 2019-08-23:
The chip supports this.
This seems to be in the documentation. It would be great.

The format of the master command for calibrating a slave (6 bytes in total):

0xF8 + 0x41 + 0x37 + 0x21 + CRC check high byte + CRC check low byte.

Correct answer: 0xF8 + 0x41 + 0x37 + 0x21 + CRC check high byte + CRC check low byte.

Error response: 0xF8 + 0xC1 + anomalous code + CRC check high byte + CRC check low byte.

It should be noted that the calibration takes from 3 to 4 seconds, after the master sends a command, if the calibration is successful, it takes 3-4 seconds to get a response from the slave.

mandulaj, 2019-10-01:
I have to still play with the calibration feature. I have heard that you can even ruin your device. I will only add it when I am confident it works well. Thanks, I will put it higher up in my priority list.

RB3rg, 2020-02-27:
I doubt this "calibration" works just like that. 
It will need a reference, a specific voltage, and current to be used as a reference.
@devicter have you ever tested this calibration?
If not, you can use node-red to test it however if I am correct, as @mandulaj said, you will ruin your device. Let us know the result.

mandulaj, 2020-04-21:
Has anyone had time to look at this?

xavipolo, 2020-04-25:
I don't think it makes sense.
As RB3rg says, the calibration function should receive a parameter that modifies the readings.
If you connect a 12W bulb, and the reading is 11W, you should be able to calibrate it to 12W or +1W

As documented the message must be 6 bytes
0xF8 + 0x41 + 0x37 + 0x21 + CRC check high byte + CRC check low byte.

0xF8: I think this is the slave address (0x01 by default)
0x41 is the calibration function
0x37 and 0x21 could be the adjustment value (data high and low bytes) signed?

morganflint, 2020-07-26:
More probably, I imagine you'll have to feed a specific voltage and current from a precision source to the module and then call the calibration function. In that case, without knowing these parameters, the calibration function is useless for us.

Nismonx, 2020-07-29:
[image]
Thought this might help...
Although I still uncertain if this is to be done with an know load or off load.

RB3rg, 2020-07-31:
0xF8 is the broadcast address for Modbus so any device will catch it.
0x41 is the function of cal.
Now comes the question, the device reads voltage and current.
Calibration needs a reference, in this case, Voltage & Current though the 0x37 0x21 results (int) 14113
I can't see 14113 being the Voltage+Current to be compared.  
Those guys are Chinese so 220V/50Hz is the standard, something around 220 should be pass in the calibration parameter.
Another point, the calibration should be float for precision (220.32), anyone can see how to pass (float)voltage + current into 2 bytes?.

Jackaro0z, 2020-08-06:
0x37 0x21 is the password. Refer to datasheet intro.

poweredscript, 2020-08-07:
I tried to calibration even though it was successful.
But when the power supply is disconnected and connected again, the MCU does not remember the calibration done.
I'm confused, what am I missing?
This is my code

`//The command format of the master to calibrate the slave is (total 6 bytes):
//               0xF8 + 0x41 + 0x37 + 0x21 + CRC check high byte + CRC check low byte.
//Correct reply: 0xF8 + 0x41 + 0x37 + 0x21 + CRC check high byte + CRC check low byte. 6
//Error Reply:   0xF8 + 0xC1 + Abnormal code + CRC check high byte + CRC check low byte. 5
bool PZEM004Tv30::setCalibration(){
    uint8_t buffer[] = {0xF8, 0x41, 0x37, 0x21, 0x00, 0x00}; // Work
	//uint8_t buffer[] = {0x01, 0x41, 0x37, 0x21, 0x00, 0x00}; // Work
    uint8_t reply[6];   

    setCRC(buffer, 6);
    _serial->write(buffer, 6);

    uint16_t length = recieve(reply, 6);

	// for(int i = 0; i < sizeof(reply); i++){
		// Serial.print("0x");
       // Serial.print(reply[i] < 16 ? "0" : "");
       // Serial.println(reply[i], HEX);
    // }
	
    if(length == 0 || length == 5){
        return false;
    }
    return true;
}`

Nismonx, 2020-08-07:
@poweredscript 

In what condition you had the PZEM when you  attempt your calibration calibration code?
In windows it always return calibration fail, have tried with/or without load.
[image]

poweredscript, 2020-08-08:
@Nismonx 

My code is adding a function to the library.
As for the program you use, I think it should have an instruction manual "the user manual of the software.txt"
If you do not use voltage and current, the program requirements cannot be calibrated.

> The user manual of the software
> 
> 1.Before use it, please click the กพrunกฟdocument to login, after login successfully, you can use the software.
> 
> 2.In all parameter setting,if the color is gray,that means it can't be amend, other color means it can be amend. 
> 
> 3.At present,this software is standalone version, not support Multimachine communication,that means if you want to 
>   use more than one module at the same time, you should set the address of each module in advance, then use the 
>   address to distinguish the termianl box when read or set the data.  
> 
> 4.Power alarm way is: the power data display become to read in the software.There is no power alarm signal output 
>   driving port in the module PCB. 
> 
> 5.Please be with caution to use the calibration function in the software!
>   The module has been calibrated at least twice with the professional equipment before delivery, so the data will
>   has problem in general.
>   If it is must be calibrated because of the defective product,the calibrationg password is: PZEM014/6
>   Calibration condition:
>   PZEM-014: VOLTAGE:220v,CURRENT:1A
>   PZEM-016: VOLTAGE:220v,CURRENT:10A

Nismonx, 2020-08-08:
@poweredscript 

I was just concerned that the calibration simply may not work and clearly misconstrued the last 2 paragraphs:

""....
   Calibration condition:
   PZEM-014: VOLTAGE:220v,CURRENT:1A
   PZEM-016: VOLTAGE:220v,CURRENT:10A  ""

I have use a 220v pressure with a 1amp load and had the same outcome (calibration failed) but its as far as I went. The picture above was a very poor choice of mine.

Thanks for pointing this out I just realise where I gone wrong , I have a PZEM-016.

morganflint, 2020-08-17:
Apparently, in the "Set parameters" option, you can see the calibration conditions for a module (but they are greyed out, so cannot be changed):
[image]

RB3rg, 2020-09-20:
> Apparently, in the "Set parameters" option, you can see the calibration conditions for a module (but they are greyed out, so cannot be changed):
> [image]
> 
> Anyway, I wouldn't calibrate it unless I had a precision AC source and load...

@morganflint can you share this PZEM014 Master?
The one I found doesn't run.
Then I will have a look into this.

morganflint, 2020-09-21:
Hello, although in the screen says "PZEM014Master", I downloaded and installed it as PZEM004TMaster:
[image]

I think I downloaded it from [the link](http://www.mediafire.com/file/lk5wukg2v5n5iry/004T_software%252Bdriver%252Buser_manual.zip/file) that can be found  at [Peacefair's Aliexpress shop product page](https://www.aliexpress.com/item/32405328185.html), but I'm enclosing it here just in case:
[004T software+driver+user manual.zip](https://github.com/mandulaj/PZEM-004T-v30/files/5257709/004T.software%2Bdriver%2Buser.manual.zip)

sirapol, 2021-05-31:
Hello , Is this issues solve or not ?
I still like your problem "The calibration failed!"

poweredscript, 2021-05-31:
> Hello , Is this issues solve or not ?
> I still like your problem "The calibration failed!"
The problem has not been resolved.

S-bilgeexuu, 2023-01-23:
Please help me?

PZEM master software Calibration password error ??? help me ??
[image]

sanj2002, 2024-11-03:
Iam using software for calibration it won't take default password help me out friends if any one did calibration through pc Software