mandulaj/PZEM-004T-v30 #15: [Arduino Nano] PZEMSoftwareSerial example

kozfelipe, 2019-11-17:
Hey guys

Im running PZEMSoftwareSerial example in my arduino nano using 115200 serial baud rate.

Heres my serial output every 2s:
```
Error reading voltage
Error reading current
Error reading power
Error reading energy
Error reading frequency
Error reading power factor

Error reading voltage
Error reading current
Error reading power
Error reading energy
Error reading frequency
Error reading power factor
```

I got my D11 and D12 setup according to example although I got only RX blinking at arduino side and TX blinking at PZEM side.

How can I diagnose this?

kozfelipe, 2019-11-17:
nevermind Ive inverted my tx/rx

closed 2019-11-17 by kozfelipe

mandulaj, 2019-11-20:
Yeah happens to the best of us :smile: Maybe this should be better clarified in the examples