mandulaj/PZEM-004T-v30 #17: pzem.setAddress() function doesn't work?

gaurav-1337lab, 2019-12-16:
Firstly, thank you very much for creating this library. It works great! Recently, I tried the ChangeAddress example with Arduino Mega and Particle's Argon board but couldn't change the address. It prints out 248 always. If I initialize the meter with say PZEM004Tv30 pzem(10, 11, 0x07);, it does work and getAddress() gives me the same.

Anything I'm doing wrong?

mandulaj, 2019-12-16:
The address 248 (0xF8) is the default, or also called broadcast address.

The PZEM addressing works like this. Each device has an address between `0x01` and `0xF7` that is saved on the device. If you have just one device on the bus, you can always address it using the broadcast address `0xF8` (This is used by default). However you should *not* use the broadcast address if you have multiple devices on the bus as all of them would try to respond to your messages. 

The function `setAddress` sends a command to the device to change its saved address. The way this is intended to be used is by configuring each of your devices individually using the example program for setting addresses. After that you can connect them all to the same bus and instantiate individual objects for each with the given addresses:

```
PZEM004Tv30 pzem1(10, 11, 0x07);
PZEM004Tv30 pzem2(10, 11, 0x08);
PZEM004Tv30 pzem3(10, 11, 0x09);
```
Do you have just one device on the bus when using the default address? 
What does the function `setAddress` return?
Try setting the address using the example program.
Let me know how this works. If you have further problems, let me know and I will look into it.

gaurav-1337lab, 2019-12-17:
Hi thanks for the quick response,

Do you have just one device on the bus when using the default address?
Yes, just one.

What does the function setAddress return?
0.

I am using the same example code provided in the library to change address. Just I am connecting over SoftwareSerial and I modified 2 lines:

uint8_t addr = 0x06;

bool ret = pzem.setAddress(addr);

Output is:
0
Current address:248

mandulaj, 2019-12-17:
One more thing, is your PZEM connected to the AC voltage? It turns out the chip does not work being powered just by the 5 volts...

gaurav-1337lab, 2019-12-17:
Oh!!! So this was the issue! I connected the mains and it started increment the device address!

gaurav-1337lab, 2019-12-17:
So I 'set' the address, then modified the program to just 'get' me the address, it again returns back 248. Shouldn't it be hardcoded?

mandulaj, 2019-12-17:
Yeah this is a bit of a quirk of the library. The getAddress function is just a getter for the internal address variable of the object. So if you initialize the object with address 0xF8, it will return 0xF8. You will have to initialize it with your new address.
I am working on changing this in the future. But as there is no way of getting the address from the device, I am still unsure how to do it.

Also, in order to explain why it does not work without the AC. The chip actually is power of it. The 5v you connect to it only serve to power to optocouplers for the RX TX communication.

You don't want to know how much time I have wasted before figuring this out :smile:

mandulaj, 2019-12-17:
I added a not about this in the Readme file. If you have any more questions, feel free to contact me.
f713ab40f0c8ece813b95a641d0c5e1535a9dbd5

closed 2019-12-17 by mandulaj

SN-8023, 2020-06-14:
> The address 248 (0xF8) is the default, or also called broadcast address.
>
> The PZEM addressing works like this. Each device has an address between `0x01` and `0xF7` that is saved on the device. If [...]

0x00 is actually the broadcast address, 0xf8 is the single node environmental address.

Quotation manual:
"The address range of the slave is 0x01 - 0xF7. The address 0x00 is used as the broadcast
address, the slave does not need to reply the master. The address 0xF8 is used as the general
address, this address can be only used in single-slave environment and can be used for calibration
etc. operation."

DAVIDBETAN2000, 2021-06-18:
tienen razón, si direcciona, es solo definir la dirección uno por uno antes de usarlos en paralelo y crear varias instancias del objeto con las mismas direcciones
PZEM004Tv30 pzem(&Serial2, 12);
PZEM004Tv30 pzem2(&Serial2, 8);
Gracias

IAmNewbe, 2023-07-12:
lgtm !