mandulaj/PZEM-004T-v30 #18: Cannot use the library on Platformio

wit03, 2020-01-23:
I've tried to import it to the project and complied with platformio core there's an error that seems to come out from SoftwareSerial library. Does anyone have any solution 

[image]

here is the error log:
`Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-----------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 1.15.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.1.1
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PZEM-004T-v30> 1.0.0
|   |-- <EspSoftwareSerial> 6.6.4
Building in release mode
Compiling .pio/build/megaatmega2560/src/main.cpp.o
Compiling .pio/build/megaatmega2560/lib2c4/EspSoftwareSerial_ID168/SoftwareSerial.cpp.o
Compiling .pio/build/megaatmega2560/libad6/PZEM-004T-v30_ID6551/PZEM004Tv30.cpp.o
Archiving .pio/build/megaatmega2560/libFrameworkArduinoVariant.a
Compiling .pio/build/megaatmega2560/FrameworkArduino/CDC.cpp.o
Indexing .pio/build/megaatmega2560/libFrameworkArduinoVariant.a
Compiling .pio/build/megaatmega2560/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/megaatmega2560/FrameworkArduino/HardwareSerial0.cpp.o
Compiling .pio/build/megaatmega2560/FrameworkArduino/HardwareSerial1.cpp.o
Compiling .pio/build/megaatmega2560/FrameworkArduino/HardwareSerial2.cpp.o
In file included from /Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.h:27:0,
                 from /Users/wit03/.platformio/lib/PZEM-004T-v30_ID6551/PZEM004Tv30.h:30,
                 from /Users/wit03/.platformio/lib/PZEM-004T-v30_ID6551/PZEM004Tv30.cpp:1:
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:210:19: error: 'Delegate' has not been declared
     void for_each(Delegate<void(T&&), ForEachArg> fun);
                   ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:210:27: error: expected ',' or '...' before '<' token
     void for_each(Delegate<void(T&&), ForEachArg> fun);
                           ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:222:31: error: 'Delegate' has not been declared
     bool for_each_rev_requeue(Delegate<bool(T&), ForEachArg> fun);
                               ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:222:39: error: expected ',' or '...' before '<' token
     bool for_each_rev_requeue(Delegate<bool(T&), ForEachArg> fun);
                                       ^
In file included from /Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.h:27:0,
                 from /Users/wit03/.platformio/lib/PZEM-004T-v30_ID6551/PZEM004Tv30.h:30,
                 from src/main.cpp:2:
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:210:19: error: 'Delegate' has not been declared
     void for_each(Delegate<void(T&&), ForEachArg> fun);
                   ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:210:27: error: expected ',' or '...' before '<' token
     void for_each(Delegate<void(T&&), ForEachArg> fun);
                           ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:222:31: error: 'Delegate' has not been declared
     bool for_each_rev_requeue(Delegate<bool(T&), ForEachArg> fun);
                               ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:222:39: error: expected ',' or '...' before '<' token
     bool for_each_rev_requeue(Delegate<bool(T&), ForEachArg> fun);
                                       ^
In file included from /Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.h:27:0,
                 from /Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:23:
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:210:19: error: 'Delegate' has not been declared
     void for_each(Delegate<void(T&&), ForEachArg> fun);
                   ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:210:27: error: expected ',' or '...' before '<' token
     void for_each(Delegate<void(T&&), ForEachArg> fun);
                           ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/circular_queue/circular_queue.h:222:31: error: 'Delegate' has not been declared
     bool for_each_rev_requeue(Delegate<bool(T&), ForEachArg> fun);
[... 259 lines cut ...]
     uint32_t curCycle = ESP.getCycleCount();
                         ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp: In static member function 'static void SoftwareSerial::rxBitSyncISR(SoftwareSerial*)':
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:509:22: error: 'ESP' was not declared in this scope
     uint32_t start = ESP.getCycleCount();
                      ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp: At global scope:
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:531:32: error: variable or field 'onReceive' declared void
 void SoftwareSerial::onReceive(Delegate<void(int available), void*> handler) {
                                ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:531:32: error: 'Delegate' was not declared in this scope
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:531:41: error: expected primary-expression before 'void'
 void SoftwareSerial::onReceive(Delegate<void(int available), void*> handler) {
                                         ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:531:62: error: expected primary-expression before 'void'
 void SoftwareSerial::onReceive(Delegate<void(int available), void*> handler) {
                                                              ^
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp: In member function 'void SoftwareSerial::perform_work()':
/Users/wit03/.platformio/lib/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:538:9: error: 'receiveHandler' was not declared in this scope
     if (receiveHandler) {
         ^
*** [.pio/build/megaatmega2560/src/main.cpp.o] Error 1
*** [.pio/build/megaatmega2560/libad6/PZEM-004T-v30_ID6551/PZEM004Tv30.cpp.o] Error 1
*** [.pio/build/megaatmega2560/lib2c4/EspSoftwareSerial_ID168/SoftwareSerial.cpp.o] Error 1
================================= [FAILED] Took 1.33 seconds =================================`

sergiocntr, 2020-01-23:
Sorry for my poor english.
1)Why to use software serial if mega has 3 hw serial.
2) remove the library that belong to esp8266 (I think) 
-> Users /wit03/.platformio/lib/EspSoftwareSerial_ID168

wit03, 2020-01-23:
Oh, I jst noticed that there's a problem with another library. Thanks for your help!

closed 2020-01-23 by wit03