mandulaj/PZEM-004T-v30 #20: Add ESP32 Support

Navok85, 2020-02-25:
Please add ESP32 support.

I changed line #25 in PZEM004Tv30.h file to:

`#if (not defined(PZEM004_NO_SWSERIAL)) && (defined(__AVR__) || defined(ESP8266)) || defined(ESP32)
`
which makes it works with hardwareserial e.g. serial2

But softwareserial still makes problems and didn't work with the EspSoftwareSerial.

clowrey, 2020-03-04:
I don't have anything to add as far as fixing the issues - but I can say I have duplicated the same basic effect. 

It works with Serial2, does not work with the EspSoftwareSerial library found here https://github.com/plerup/espsoftwareserial

This might be tricky to debug - not sure exactly where to start, my board the TTGO ESP32 TFT with LCD screen had no available extra serial as the Serial2 port is used by the built in LCD screen. I switched to a different board with the hardware serial pins 16,17 available and it works with the hardware serial example.

mudassar791, 2020-04-29:
can you upload ready made example with esp32 & esp8266 in next release?

mandulaj, 2021-07-16:
I am aware of the Software Serial bug in ESP32 and I don't have a solution. However I think that since the ESP32 has 3 HW serial ports which can be configured to use any GPIO pin, there isn't really any need for a Software Serial solution. 

If there is a requirement to use the Software Serial, I might consider looking into it in the future. But for now, I think its better to just not support it on the ESP32 platform.

ASMotionLab, 2021-07-16:
Yes, you no longer need to use SoftwareSerial on the ESP32, even to address multiple pzems thanks to the new updates 🎉
Just remap the HardwareSerial pins so some free unused GPIOs and it works great. Not sure if the library does this remap for you, but you can do it brute force by editing this file:
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\HardwareSerial.cpp
Then close & reopen Arduino & recompile

mandulaj, 2021-07-16:
OK, The newest code has now been merged into master. Soon I will update the PlatformIO and Arduino Library manager library. Here is the the new way of defining HW Serial on ESP32s:

```c++
// RX=16, TX=17
PZEM004Tv30 pzem(Serial2, 16, 17);
```
If there isn't anything else, I will close this issue.

closed 2021-07-16 by mandulaj

rohanpatil-rc, 2023-09-08:
i am having issues (esp32)
the code get compiled and uploaded
but serial monitor reports -error reading
i have tried swaping rx and tx but still not getting any outuput 
the pzem ttl only one led blinks  
change the baud rate to 9600

mandulaj, 2023-09-08:
@rohanpatil-rc You will have to give me more info in order to help you. Which pins are you using on the ESP32, can you share the code? Is the PZEM powered by AC?  Could you share the code or library that does work? Are you sure you are using a PZEM 0004T version 3.0?

rohanpatil-rc, 2023-09-08:
same code in example 

serial monitor - Error reading voltage
and only one led of pzem ttl is blinking 
tried to swap rx and tx but no luck.

i have esp32 wroom devkit 1

code-
#include <PZEM004Tv30.h>

/* Hardware Serial2 is only available on certain boards.
 * For example the Arduino MEGA 2560
*/
#if defined(ESP32)
PZEM004Tv30 pzem(Serial2, 16, 17);
#else
PZEM004Tv30 pzem(Serial2);
#endif

void setup() {
    Serial.begin(9600);

    // Uncomment in order to reset the internal energy counter
     //pzem.resetEnergy();
}

void loop() {
        
    Serial.print("CUSTOM:");
    //Serial.println(pzem.readAddress(), HEX);

    // Read the data from the sensor
    float voltage = pzem.voltage();
    float current = pzem.current();
    float power = pzem.power();
    float energy = pzem.energy();
    float frequency = pzem.frequency();
    float pf = pzem.pf();

    // Check if the data is valid
    if( !isnan(voltage))
    {
        Serial.println("Error reading voltage");
    } else if (!isnan(current)) {
        Serial.println("Error reading current");
    } else if (!isnan(power)) {
        Serial.println("Error reading power");
    } else if (!isnan(energy)) {
        Serial.println("Error reading energy");
    } else if (!isnan(frequency)) {
        Serial.println("Error reading frequency");
    } else if (!isnan(pf)) {
        Serial.println("Error reading power factor");
    } else {

        // Print the values to the Serial console
        Serial.print("Voltage: ");      Serial.print(voltage);      Serial.println("V");
        Serial.print("Current: ");      Serial.print(current);      Serial.println("A");
        Serial.print("Power: ");        Serial.print(power);        Serial.println("W");
        Serial.print("Energy: ");       Serial.print(energy);     Serial.println("kWh");
        Serial.print("Frequency: ");    Serial.print(frequency); Serial.println("Hz");
        Serial.print("PF: ");           Serial.println(pf);

    }

    Serial.println();
    delay(2000);
}

rohanpatil-rc, 2023-09-09:
> @rohanpatil-rc You will have to give me more info in order to help you. Which pins are you using on the ESP32, can you share the code? Is the PZEM powered by AC? Could you share the code or library that does work? Are you sure you are using a PZEM 0004T version 3.0?

i have delete my post with the code that is was having issues.
it was not the code it was the gnd pin trace of esp32 that was broke 
its working now 
thanks for your library