mandulaj/PZEM-004T-v30 #21: Software Serial

SK-sep, 2020-04-08:


mandulaj, 2020-04-21:
Hi, what exactly is this issues? Could you please provide more information. Thanks

pablovesga, 2020-09-15:
Software serial doesn't function with PZEM-004T-v30

Impossible connection with bluetooth. Pins 2 and 3 don't function to include PZEM

Please Help me

<code>
#include <SoftwareSerial.h> //Librería que permite establecer comunicación serie en otros pins
#include <PZEM004Tv30.h> 

SoftwareSerial BT(2, 3); //2 RX, 3 TX.
PZEM004Tv30 vat(4, 5); //4 RX, 5 TX.

float v=0;
float c=0;
float p=0;
float e=0;
float f=0;
float pf=0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600); //Abrimos la comunicación serie con el PC y establecemos velocidad
  BT.begin(9600); //Velocidad del puerto del módulo Bluetooth
  pinMode(13, OUTPUT);
  pinMode(12, OUTPUT);
  pinMode(10, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:

  if (BT.available())
  {
    digitalWrite(13, HIGH);
    delay(200);
    char dato = BT.read();
    switch (dato){
      case 'a': digitalWrite(10, HIGH);
      break;
      case 'b': digitalWrite(12, HIGH);
      break;
      case 'c': digitalWrite(10, HIGH);
      break;
      case 'd': digitalWrite(13, LOW);
      digitalWrite(12, LOW);
      digitalWrite(10, LOW);
      break;
      case 'e': BT.println("Juan Pablo");
      break;
      case 'f': BT.println("Diego Alberto");
      break;
      case 'g': BT.println("Leidy");
      break;
      case 'h': BT.println("Astrid");
      break;
      case 'i': v = vat.voltage();
        if( !isnan(v) ){
          BT.print("Voltaje: "); BT.print(v); BT.print("V - ");
          c = vat.current();
          BT.print("Corriente: "); BT.print(c); BT.print("A - ");
          p = vat.power();
          BT.print("Potencia: "); BT.print(p); BT.print(" W - ");
          e = vat.energy();
          BT.print("Energía: "); BT.print(e,3); BT.print(" kWh - ");
          f = vat.frequency();
          BT.print("Frecuencia: "); BT.print(f, 1); BT.print(" Hz - ");
          pf = vat.pf();
          BT.print("PF: "); Serial.println(pf);
        } else {
          BT.println("Error en lectura de voltaje");
        }
        BT.println();
        vat.resetEnergy();
        delay(2000);
      break;
    }
    digitalWrite(13, LOW);
  }
  BT.flush();
}
</code>

mandulaj, 2021-07-16:
This issue should be resolved with the new library version. I will close it as we have this issue duplicated multiple times.

closed 2021-07-16 by mandulaj