mandulaj/PZEM-004T-v30 #23: always getting error reading message with default example

mcfly4711, 2020-04-13:
Hi I am using an Arduino Uno and tried to connect the PZEM-004-v30 RX -->D11 and TX --> D12.
when I compiled and uploaded the default example I gets always the following message in the serial monitor.
Error reading voltage
Error reading current
Error reading power
Error reading energy
Error reading frequency
Error reading power factor

code which i have used:
#include <PZEM004Tv30.h>

/* Use software serial for the PZEM
 * Pin 11 Rx (Connects to the Tx pin on the PZEM)
 * Pin 12 Tx (Connects to the Rx pin on the PZEM)
*/
PZEM004Tv30 pzem(11, 12);

void setup() {
  Serial.begin(115200);
}

void loop() {
    float voltage = pzem.voltage();
    if( !isnan(voltage) ){
        Serial.print("Voltage: "); Serial.print(voltage); Serial.println("V");
    } else {
        Serial.println("Error reading voltage");
    }

    float current = pzem.current();
    if( !isnan(current) ){
        Serial.print("Current: "); Serial.print(current); Serial.println("A");
    } else {
        Serial.println("Error reading current");
    }

    float power = pzem.power();
    if( !isnan(power) ){
        Serial.print("Power: "); Serial.print(power); Serial.println("W");
    } else {
        Serial.println("Error reading power");
    }

    float energy = pzem.energy();
    if( !isnan(energy) ){
        Serial.print("Energy: "); Serial.print(energy,3); Serial.println("kWh");
    } else {
        Serial.println("Error reading energy");
    }

    float frequency = pzem.frequency();
    if( !isnan(frequency) ){
        Serial.print("Frequency: "); Serial.print(frequency, 1); Serial.println("Hz");
    } else {
        Serial.println("Error reading frequency");
    }

    float pf = pzem.pf();
    if( !isnan(pf) ){
        Serial.print("PF: "); Serial.println(pf);
    } else {
        Serial.println("Error reading power factor");
    }

    Serial.println();
    delay(2000);
}

any Idea what I am doing wrong?

mandulaj, 2020-04-21:
Hey, did you make sure to connect the module to the AC? It is actually not powered by the 5V you give it from the Arduino. It uses the AC to power itself. The 5V only power the opto-couplers which isolate your Arduino form the mains....

mcfly4711, 2020-04-21:
Yes power 230 V is also connected but not the CT. 
It seams that on  RX there is no signal. No led on the board is flashing.

mcfly4711, 2020-04-21:
Thx for your email.Yes Ac 230V is connected but not CT.It seams for me that there is no signal pn RX port. No Led flashing like on TX.

mandulaj, 2020-04-21:
Hmm. that is strange. The CT not connected should not be causing any issues (you will just read 0 current). 

Could you try this, swap the RX and TX wires. Maybe you have them the other way around

mcfly4711, 2020-04-21:
I will test in the evening:-)

mcfly4711, 2020-04-21:
No changes. I checked connection snd changed TX and RX.  But it was wired correctly.Am 21.04.2020 15:49 schrieb ralf.kallfass@my-webspace.de:I will test in the evening:-)

mcfly4711, 2020-10-23:
Flapping connection :-( thx for your support

closed 2020-10-23 by mcfly4711

Jzstay, 2023-12-02:
did  you resolve it?

Jemeeeeer, 2024-04-02:
im having the exact same problem. any update on how you resolved the issue?

Jemeeeeer, 2024-04-02:
> Flapping connection :-( thx for your support

what was the issue?

mcfly4711, 2024-04-02:
No I could not fix it

Chileshe1, 2024-08-20:
I'm facing the same challenge right now

mandulaj, 2024-08-20:
@Chileshe1 Did you check:
- PZEM is powerd by the AC and the DC!