mandulaj/PZEM-004T-v30 #24: Problem with integrating your lib example

Groofertje, 2020-04-25:
Hey, i'am using a ESP32 and coding in Visual Studio Code.

I copied your example SoftwareSerial but getting a compiling errors

no instance of constructor "PZEM004Tv30::PZEM004Tv30" matches the argument list -- argument types are: (int, int)

and

invalid conversion from 'int' to 'HardwareSerial*' [-fpermissive]

It's happend at the line PZEM004Tv30 pzem(11, 12);

/* Use software serial for the PZEM
 * Pin 11 Rx (Connects to the Tx pin on the PZEM)
 * Pin 12 Tx (Connects to the Rx pin on the PZEM)
*/
PZEM004Tv30 pzem(11, 12);        <======================

and giving the next statement;

D:\_Dev\_MyLib\PZEM-004T-v30/PZEM004Tv30.h:65:5: note:   initializing argument 1 of 'PZEM004Tv30::PZEM004Tv30(HardwareSerial*, uint8_t)'
     PZEM004Tv30(HardwareSerial* port, uint8_t addr=PZEM_DEFAULT_ADDR);

sergiocntr, 2020-04-25:
Hi, 
there something wrong with your specific board.
As first workaround comment lines  47 and 49 in PZEM004Tv30.h .
This will force software serial.
`// #define PZEM004_NO_SWSERIAL
//#if (not defined(PZEM004_NO_SWSERIAL)) && (defined(__AVR__) || defined(ESP8266) && (not defined(ESP32)))
#define PZEM004_SOFTSERIAL
//#endif`

More deeply ,a  verbose build will notice the defined board es.
> '-D' 'ESP8266' '-D' 'ARDUINO_ARCH_ESP8266' '-D' 'ARDUINO_ESP8266_ESP01' 
this is in my case the ESP01 from ESP8266 family.
With this information you can adapt the if statement .

Groofertje, 2020-04-26:
Tx for the answer, but the problem remains.
i changed the As first workaround comment lines 47 and 49 in PZEM004Tv30.h .

// #define PZEM004_NO_SWSERIAL
// #if (not defined(PZEM004_NO_SWSERIAL)) && (defined(__AVR__) || defined(ESP8266))
#define PZEM004_SOFTSERIAL
// #endif

but keep getting the compiling error in Visual Code

for your information i give al what i found.

PLATFORM: Espressif 32 1.12.0 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PZEM-004T-v30> 1.0.0
|   |-- <EspSoftwareSerial> 6.8.1
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
Generating partitions .pio\build\esp32doit-devkit-v1\partitions.bin
Compiling .pio\build\esp32doit-devkit-v1\libfd4\EspSoftwareSerial_ID168\SoftwareSerial.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libe67\PZEM-004T-v30\PZEM004Tv30.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libd7d\WiFi\ETH.cpp.o
src\main.cpp:25:23: error: invalid conversion from 'int' to 'HardwareSerial*' [-fpermissive]
 PZEM004Tv30 pzem(25,18);
                       ^
In file included from src\main.cpp:14:0:
D:\_Dev\_MyLib\PZEM-004T-v30/PZEM004Tv30.h:65:5: note:   initializing argument 1 of 'PZEM004Tv30::PZEM004Tv30(HardwareSerial*, uint8_t)'
     PZEM004Tv30(HardwareSerial* port, uint8_t addr=PZEM_DEFAULT_ADDR);
     ^
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1

Groofertje, 2020-04-26:
srr i closed it by error :(

Groofertje, 2020-04-26:
in verbose mode i got this
src\main.cpp:25:23: error: invalid conversion from 'int' to 'HardwareSerial*' [-fpermissive]
 PZEM004Tv30 pzem(25,18);
                       ^
In file included from src\main.cpp:14:0:
xtensa-esp32-elf-g++ -o .pio\build\esp32doit-devkit-v1\FrameworkArduino\IPv6Address.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -g3 -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions -Werror=reorder -DPLATFORMIO=40301 -DARDUINO_ESP32_DEV -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DARDUINO=10805 -DARDUINO_ARCH_ESP32 -DARDUINO_VARIANT=\"doitESP32devkitV1\" "-DARDUINO_BOARD=\"DOIT ESP32 DEVKID:\_Dev\_MyLib\PZEM-004T-v30/PZEM004Tv30.h:65:5: note:   initializing argument 1 of 'PZEM004Tv30::PZEM004Tv30(HardwareSerial*, uint8_t)'
T V1\"" -IC:\Users\guy_s\.pla     PZEM004Tv30(HardwareSerial* port, uint8_t addr=PZEM_DEFAULT_ADDR);
tf     ^
orm*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1

sergiocntr, 2020-04-26:
Strange, it works for me. Try again.
I'm not the maintener but it seems that the use of  hardware serial is indicated for the esp32 .
Also check that you have the latest version of the library from the github repository,  you miss the ESP32 part in the if statement.

Groofertje, 2020-04-26:
tx for respons

1) Also check that you have the latest version of the library from the github repository
Yes i did, i even downloaded the git reposytory to view if there where changes.

2) you miss the ESP32 part in the if statement.
??? Now i can't follow   :)

Groofertje, 2020-04-26:
Ok i resolved the problem, it's was complete my fault.
in my previous answer i noted that i had even downloaded the github repository to my computer. 
and that was the problem.
I putted it into mylib folder and vscode pointed to this folder and not the version i downloaded via vscode library's
I moved it to another folder and now the code compiles normaly.

But i don't know why he give a error with the downloaded github and NOT with the version that i downloaded via PIOhome, (VS code)

closed 2020-05-03 by mandulaj