mandulaj/PZEM-004T-v30 #26: Compatibility with Nodemcu ESP32 board?

Dhruv3872, 2020-05-05:
I have used the library to work with Arduino UNO and it works fine.
With my Nodemcu-32S board, I am not able to get anything on the serial monitor when I run "PZEMHardSerial" example with "&Serial" instead of "&Serial3".
Is your library compatible with Nodemcu-32S module?
If not, is there a way to make it compatible?

Thanks.

mandulaj, 2020-05-06:
Hi @Dhruv3872, I have the library running on an ESP32 with hardware serial no problem. Have you tried any of the common issue solutions: 
* Make sure the PZEMs are connected to main voltage
* Try swapping RX-TX wires (Serial RX must be connected to TX, I get them confused myself sometimes) Good indicaton is to look at the flashing LEDs on the PZEMs, Are both flashing or just one?
* Have you connected to the correct HW Serial pins/ Selected the correct HW serial

thanawat14, 2020-05-25:
@mandulaj I have the same problem.. "Error reading ..." 
I use &Serial2 and connect Tx/Rx to 16,17 pin of esp32. I'm quite sure I connect to Tx-Rx correctly but result is the same.. Btw, what i can see is only Rx led is blinking.

ktotomskru, 2020-05-27:
> @mandulaj I have the same problem.. "Error reading ..."
> I use &Serial2 and connect Tx/Rx to 16,17 pin of esp32. I'm quite sure I connect to Tx-Rx correctly but result is the same.. Btw, what i can see is only Rx led is blinking.

I'm getting exactly the same results with my pzem-004t from ali and esp32. RX on Pzem flashes during poll, but TX does not. Thus "error reading voltage/current/etc." goes from esp32 to usb-serial. 
I tried different serial connection, hardware on pins GPIO1+GPIO3, GPIO16+GPIO17 and serial on different other pins. Still no luck. 
By any chance, didn't you manage to solve it somehow?

mandulaj, 2020-07-28:
Yeah this is a very common problem. It is already in the Readme however if you have any ideas how to make it more clear, let me know. I am closing this issue for now.

closed 2020-07-28 by mandulaj

Jemeeeeer, 2024-03-25:
Hi, my pzem004t datasheet tells me that the command format for reading the sensor values is different from the sendcmd8 in your library. is it the explanation why i get nan values in my serial monitor?

mandulaj, 2024-03-26:
@Jemeeeeer This library is for the PZEM004T v3.0. If you have the PZEM004T, this library will not work and you should use https://github.com/olehs/PZEM004T instead.

mandulaj, 2024-03-26:
Could you share the datasheet you have?

mandulaj, 2024-03-27:
Yep thats interesting, I can't find the datasheet I used to implement this anymore. I will have to have a closer look at this. But strange that something like that hasn't been observed earlier. Besides on my PZEMs it worked fine. Have you tried swapping the byte order?

Jemeeeeer, 2024-03-27:
i have tried changing the crc byte order in your library by modifying it. it can transmit a command format however, there is no available bytes to read in the response buffer. what seems to be the problem?

mandulaj, 2024-03-27:
Do both LEDs flash on the board?

Jemeeeeer, 2024-03-27:
only the TX LED is blinking,

Jemeeeeer, 2024-03-27:
do u have any idea why my pzem004tv30 rxx pin not blinking? both of the rx and tx pins are connected to a voltage divider so to make it 3.3 V. pzem004tv30 is powered with a 5v external dc supply, while esp32 is powered with microusb connected to pc. im using pins 3, 1 for rxtx serial 0. i still dont know why it isnt receiving data.

Jemeeeeer, 2024-04-02:
up

mandulaj, 2024-04-02:
Do you have the 240V AC line attached? That is where the PZEM chip actually gets the power form. The 5V is only to power the opto-couplers.