mandulaj/PZEM-004T-v30 #29: Multiple PZEMS cannot be used using same library

alihassan2289, 2020-06-22:
Dear good day,

Appreciate your efforts, can you please update the library to use ip function to add multiple pzem v3 devices in the same code. 

Thanks

Renato-REDS, 2020-07-24:
Hello, thank you very much for sharing this library with us. It would be very important if we were able to create a library and be able to use multiple PZEM on the same network. thank you.

efakostya, 2020-07-27:
PZEM004Tv30* pzem[] = {
new PZEM004Tv30(D1, D2),
new PZEM004Tv30(D3, D4),
new PZEM004Tv30(D5, D6)
};

for (size_t i = 0; i < 3; i++){
v[i] = pzem[i]->voltage();
...
}

Renato-REDS, 2020-07-27:
Hi  @Efakostya,
Many thanks for the reply.

In my case I already have the hardware ready, and the communication of the PZEM's are in parallel, I cannot change the pins.

morganflint, 2020-08-10:
I managed to read several sensors connected in parallel to the same software serial port. The first approach, based on the example that is presented in "readme.md", was to create two instances with different names, same port, and different address:
```
#include <PZEM004Tv30.h>

/* Use software serial for the PZEM
 * Pin 11 Rx (Connects to the Tx pin on both PZEMs)
 * Pin 12 Tx (Connects to the Rx pin on both PZEMs)
 *Two instances, one for each address
*/
PZEM004Tv30 pzem1(11,12, 0x01);
PZEM004Tv30 pzem2(11,12, 0x02);

void setup() {
  Serial.begin(115200);
}

void loop() {
    // 1st sensor
    Serial.println("==================================================");
    Serial.println("Sensor 1");
    float voltage = pzem1.voltage();
    if( !isnan(voltage) ){
        Serial.print("Voltage: "); Serial.print(voltage); Serial.println("V");
    } else {
        Serial.println("Error reading voltage");
    }

    float current = pzem1.current();
    if( !isnan(current) ){
        Serial.print("Current: "); Serial.print(current); Serial.println("A");
    } else {
        Serial.println("Error reading current");
    }

    float power = pzem1.power();
    if( !isnan(power) ){
        Serial.print("Power: "); Serial.print(power); Serial.println("W");
    } else {
        Serial.println("Error reading power");
    }

    float energy = pzem1.energy();
    if( !isnan(energy) ){
        Serial.print("Energy: "); Serial.print(energy,3); Serial.println("kWh");
    } else {
        Serial.println("Error reading energy");
    }

    float frequency = pzem1.frequency();
    if( !isnan(frequency) ){
        Serial.print("Frequency: "); Serial.print(frequency, 1); Serial.println("Hz");
    } else {
        Serial.println("Error reading frequency");
    }

    float pf = pzem1.pf();
    if( !isnan(pf) ){
        Serial.print("PF: "); Serial.println(pf);
    } else {
        Serial.println("Error reading power factor");
    }
    Serial.println();
    delay(1000);

    // 2nd sensor
    Serial.println("==================================================");
    Serial.println("Sensor 2");
    voltage = pzem2.voltage();
    if( !isnan(voltage) ){
        Serial.print("Voltage: "); Serial.print(voltage); Serial.println("V");
[... 137 lines cut ...]
Current: 0.04A
Power: 0.50W
Energy: 0.026kWh
Frequency: 50.0Hz
PF: 0.06

SENSOR No. 2
Voltage: 236.90V
Current: 0.07A
Power: 0.90W
Energy: 0.184kWh
Frequency: 50.0Hz
PF: 0.05

SENSOR No. 3
Error reading voltage
Error reading current
Error reading power
Error reading energy
Error reading frequency
Error reading power factor
```
The errors for sensor No 3 are due to the fact I haven't connected it yet ;-)

Hope it helps!

Renato-REDS, 2020-08-11:
Hi @morganflint,

Thank you! Do me a lot! Congratulations !!!

mandulaj, 2020-08-19:
Thanks @morganflint. Do you guys think the instructions should be placed somewhere more visible since this is a recurring issue people are coming across. If anyone has time and strength to describe this in the wiki or something, feel free to do so!

closed 2020-08-19 by mandulaj