mandulaj/PZEM-004T-v30 #34: problems with wemos d1 and PZEM-004T

yumupinglan, 2020-08-18:
After running a few minutes the system will crush. Error messages:
`Exception 0: Illegal instruction
PC: 0x402021e4: circular_queue ::push(unsigned int&&) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/atomic_base.h line 420
EXCVADDR: 0x00000000

Decoding stack results
0x4010079c: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 166
0x40100e2c: check_poison_block(umm_block*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_poison.c line 95
0x40100cd1: umm_free_core(void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 351
0x401006d8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 137
0x401006d8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 137
0x401006d8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 137
0x401006d8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 137
0x40204768: loop_task(ETSEvent*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_features.h line 92
0x4020dee4: sys_timeout_LWIP2 at core/timeouts.c line 304
0x4020df57: tcpip_tcp_timer at core/timeouts.c line 153
0x40204784: loop_task(ETSEvent*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 208
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x4020d6b2: glue2esp_linkoutput at glue-esp/lwip-esp.c line 241
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401003b0: SoftwareSerial::rxBitISR(SoftwareSerial*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/unique_ptr.h line 420
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401007b8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 174
0x4010079c: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 166
0x401003b0: SoftwareSerial::rxBitISR(SoftwareSerial*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/unique_ptr.h line 420
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401006d8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 137
0x40100569: millis() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 193
0x4010054c: millis() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 175
0x402021dc: std::_Function_handler   >::_M_invoke(const std::_Any_data &, unsigned int &&) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 423
0x40201ca8: circular_queue ::for_each(Delegate  const&) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/atomic_base.h line 122
0x401001ce: SoftwareSerial::writePeriod(unsigned int, unsigned int, bool) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 263
0x40201f88: SoftwareSerial::rxBits() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/std_function.h line 303
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x402021d0: std::_Function_handler   >::_M_invoke(const std::_Any_data &, unsigned int &&) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 423
0x4020483b: __yield() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_features.h line 92
0x402015fa: PZEM004Tv30::recieve(unsigned char*, unsigned short) at /home/willieyu/Arduino/libraries/PZEM-004T-v30-master/PZEM004Tv30.cpp line 469
0x402016aa: PZEM004Tv30::sendCmd8(unsigned char, unsigned short, unsigned short, bool, unsigned short) at /home/willieyu/Arduino/libraries/PZEM-004T-v30-master/PZEM004Tv30.cpp line 246
0x40201752: PZEM004Tv30::updateValues() at /home/willieyu/Arduino/libraries/PZEM-004T-v30-master/PZEM004Tv30.cpp line 377
0x402018dd: PZEM004Tv30::voltage() at /home/willieyu/Arduino/libraries/PZEM-004T-v30-master/PZEM004Tv30.cpp line 130
0x40201077: getPzem() at /home/willieyu/Arduino/PZEMSoftwareSerial/PZEMSoftwareSerial.ino line 119
0x40201374: loop() at /home/willieyu/Arduino/PZEMSoftwareSerial/PZEMSoftwareSerial.ino line 78
0x40204928: loop_wrapper() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 197` 

- Wemos D5 to PZEM Tx
- Wemos D6  to PZEM Rx
- Wemos 5V to  PZEM 5V
- Wemos GND to  PZEM GND

Sketch:
`#include <PZEM004Tv30.h>
#include <ESP8266WiFi.h>

/* Use software serial for the PZEM
 * Pin 11 Rx (Connects to the Tx pin on the PZEM)
 * Pin 12 Tx (Connects to the Rx pin on the PZEM)
*/
PZEM004Tv30 pzem(D5, D6);

#ifndef APSSID
#define APSSID "willie"
#define APPSK  "09yuwei@ahu"
#endif

const char* ssid     = APSSID;
const char* password = APPSK;
[... 128 lines cut ...]
        Serial.print("Energy: "); Serial.print(energy,3); Serial.println("kWh");
//        result += "{name:'energy', value:"+String(energy)+", unit:'kWh'},";
    } else {
        Serial.println("Error reading energy");
    }

    float frequency = pzem.frequency();
    if( !isnan(frequency) ){
        Serial.print("Frequency: "); Serial.print(frequency, 1); Serial.println("Hz");
//        result += "{name:'frequency', value:"+String(frequency)+", unit:'Hz'},";
    } else {
        Serial.println("Error reading frequency");
    }

    float pf = pzem.pf();
    if( !isnan(pf) ){
        Serial.print("PF: "); Serial.println(pf);
//        result += "{name:'pf', value:"+String(pf)+", unit:''},";
    } else {
        Serial.println("Error reading power factor");
    }

//    result += "]";
//    return result;
}`

mandulaj, 2021-07-16:
Hi sorry,  I was unable to reproduce this with the new rel1.1 version. Could you try again?

mandulaj, 2021-07-16:
I will close this issue for now. In case you still have the problem even with the new release version, please open a new issue.

closed 2021-07-16 by mandulaj