mandulaj/PZEM-004T-v30 #36: Pzem reads 2-way?

Tpro4391, 2020-10-02:
Other electronic meters often distinguish the current direction.
Is there any way pzem can distinguish the current direction?
Because I want to have kWh import and kWh export
please help!

popeye2019, 2020-10-02:
Hi, if you know how tell me, I'm interrested.

luckywang95, 2020-10-04:
https://forum.arduino.cc/index.php?topic=621693.15

Based on this post, it doesn't seem like it is measuring 2 ways. Quite sad.

mandulaj, 2020-10-19:
Unfortunately, there is no way to measure the phase between the voltage and current directly using the PZEM. The power factor is the cosine of the phase and thus doesn't contain information which way the power is flowing. This is why I decided to build my own power meter which will allow the direction to be determined.

mandulaj, 2021-01-23:
Duplicate of #2

closed 2021-01-23 by mandulaj

ThomasDiebel, 2021-03-10:
> Unfortunately, there is no way to measure the phase between the voltage and current directly using the PZEM. The power factor is the cosine of the phase and thus doesn't contain information which [...]

Thanks for the examples. Got any results on building?