mandulaj/PZEM-004T-v30 #37: Refresh rate

Alrik78, 2020-10-28:
Hi,
First i would like to thank you for the work done on library.
do you know how to speed up the refresh rate of measurements?
i did some testing with a n Arduino Mega and a 2x16 display to find out that the rate is around 1 sec or more.
it did not matther if i disable serial print statements.

do you have an idea?

Regards, Alrik

mandulaj, 2020-10-31:
Hey @Alrik78, I am glad you like the library. One thing you can play with is the UPDATE_TIME in the library (you will have to change that in PZEM004Tv30.cpp. By default, this is set to 200 ms meaning, it will sample the PZEMs at most every 200ms. 

Basically, if you call the various get voltage/current/power etc methods, they really only return a cached value (its more efficient). This cached value however updates, if there is more than 200ms between the individual calls (see `updateValues` function) If you decrease it, you will be able to sample faster, however a words or warning, 
A) I am not sure what is the internal sampling frequency of the PZEM, 200ms was a safe value that still allows quite high sampling resolution. I remember reading something about it but I can't find it in the datasheet now. 
B) You might start hitting the limit of the MODBUS protocol. The PZEMs only use 9600 baud. Given the transaction is about 33 bytes = 264 bits, you can only do at best 36 transactions per second ~ 28ms between transactions (not counting overhead!!!!) So sure, you can try decreasing the UPDATE_TIME, but at 28ms (most likely much sooner thought) you will definitely hit the hard limit.

Also another note, the `updateValues` updates all the values provided by the chip. Hence the transaction is quite large (but in long run overhead small) If you just need to read one property, say the voltage, you can optimize the code to only retrieve the one register, say voltage. But you would have to write your custom code to do that plust as I said earlier, I am not sure it will bring you any benefit if the PZEM internally samples say every 100ms.

mandulaj, 2021-07-16:
I will close this as an old issue.

closed 2021-07-16 by mandulaj