mandulaj/PZEM-004T-v30 #39: compile errors with MKR 1010

nmsneves, 2020-11-05:
hi, i've been using the library very well on an arduino uno, but today i decided to compile it for an arduino MKR1010, without sucess. 
is this board compatible? i tried to use both hardware and software serial. 
the errors are the following:

`In file included from C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.8\cores\arduino/delay.h:23:0,
                 from C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.8\cores\arduino/Arduino.h:81,
                 from C:\Users\\MEOCloud\Arduino\libraries\PZEM-004T-v30-master\PZEM004Tv30.h:41,
                 from C:\Users\\MEOCloud\Arduino\libraries\PZEM-004T-v30-master\PZEM004Tv30.cpp:23:
C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.8\variants\mkrwifi1010/variant.h:222:37: 

error: conflicting declaration 'HardwareSerial SerialUSB'
 #define Serial                      SerialUSB

                                     ^
C:\Users\\\Arduino\libraries\PZEM-004T-v30-master\PZEM004Tv30.cpp:54:23: note: in expansion of macro 'Serial'
 extern HardwareSerial Serial;
                       ^~~~~~
In file included from C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.8\cores\arduino/Arduino.h:139:0,
                 from C:\Users\\MEOCloud\Arduino\libraries\PZEM-004T-v30-master\PZEM004Tv30.h:41,
                 from C:\Users\\MEOCloud\Arduino\libraries\PZEM-004T-v30-master\PZEM004Tv30.cpp:23:
C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.8\cores\arduino/USB/CDC.h:158:16: note: previous declaration as 'Serial_ SerialUSB'
 extern Serial_ SerialUSB;
                ^~~~~~~~~
C:\Users\\\Arduino\libraries\PZEM-004T-v30-master\PZEM004Tv30.cpp: In destructor 'PZEM004Tv30::~PZEM004Tv30()':
C:\Users\\MEOCloud\Arduino\libraries\PZEM-004T-v30-master\PZEM004Tv30.cpp:119:22: warning: deleting object of abstract class type 'Stream' which has non-virtual destructor will cause undefined behavior [-Wdelete-non-virtual-dtor]
         delete this->_serial;
                      ^~~~~~~
exit status 1
Erro ao compilar para a placa Arduino MKR WiFi 1010.
`

mandulaj, 2020-11-05:
Hey @nmsneves! I am sorry, I don't have an Arduino MKR 1010 so I have never tested the library with it. It looks like there might be and issue with the Hardware serial declaration.... Seems like the API is not 100% compatible between the various Arduino boards. I don't know if I will have time to look into this but I will mark it as something that should be fixed so that at least the common Arduino boards are supported.

Thanks.

nmsneves, 2020-11-06:
hi @mandulaj , thanks for marking this to be fixed. And like you say, it really should be a problem with the hardware serial definition. I could make it compile and run today, just by replacing every call involving the serial port, by a call to "Serial1", and modifying the constructor

mandulaj, 2020-11-06:
@nmsneves could you perhaps create a pull request with your changes? I would need more other people to test it  since as I said, I don't have the Arduino at hand... Thanks 😊

nmsneves, 2020-11-06:
> 
> 
> @nmsneves could you perhaps create a pull request with your changes? I would need more other people to test it since as I said, I don't have the Arduino at hand... Thanks 😊

... i would, but i'm not familiar with github. give me a while to look how to do it

seramonte, 2020-11-07:
I have same problem with DUE. The following changes allow program to compile. But still no output.  

@PZEM004Tv30.h:67
`    //PZEM004Tv30(HardwareSerial* port, uint8_t addr=PZEM_DEFAULT_ADDR);
    PZEM004Tv30(USARTClass* port, uint8_t addr=PZEM_DEFAULT_ADDR);   // added for DUE`

@PZEM004Tv30.cpp:55
`//extern HardwareSerial Serial;   // commented out for DUE`

@PZEM004Tv30.cpp:102
`//PZEM004Tv30::PZEM004Tv30(HardwareSerial* port, uint8_t addr)
PZEM004Tv30::PZEM004Tv30(USARTClass* port, uint8_t addr)   // for DUE`

nmsneves, 2020-11-08:
> u

Hi @seramonte , the changes I did to have the library to compile and run on my MKR1010 were these:

like you, I comment this line:
`//extern HardwareSerial Serial;
`
modified the hardware serial constructor:
```
//PZEM004Tv30::PZEM004Tv30(HardwareSerial* port, uint8_t addr)
PZEM004Tv30::PZEM004Tv30(uint8_t addr)
{
    /*
    port->begin(PZEM_BAUD_RATE);
    this->_serial = port;
    this->_isSoft = false;
   */
    init(addr);
}
```
and every occurrence of `_serial->` was replaced by `Serial1.`

in the examples code, the only change I did to have it running was this:
```
#define PZEM_DEFAULT_ADDR    0xF8

PZEM004Tv30 pzem(PZEM_DEFAULT_ADDR);
```

Fox2829, 2020-11-10:
@nmsneves, could you clarify. 

You changed in the file PZEM004Tv30.cpp:
1.  `//extern HardwareSerial Serial;`

2. `//PZEM004Tv30::PZEM004Tv30(HardwareSerial* port, uint8_t addr)`
`PZEM004Tv30::PZEM004Tv30(uint8_t addr)`
`{`
  `  /*`
 `   port->begin(PZEM_BAUD_RATE);`
  ` this->_serial = port;`
    `this->_isSoft = false;`
  ` */`
  `  init(addr);`
`}`

3. In four places 
`// _serial->write(sendBuffer, 8); // send frame`
`Serial1.write(sendBuffer, 8); // send frame`
or
` _Serial1->write(sendBuffer, 8); // send frame`
or somehow differently?

4. And You changed in the file PZEMHardSerial.ino:
`//PZEM004Tv30 pzem(&Serial3);`
`#define PZEM_DEFAULT_ADDR    0xF8`
`PZEM004Tv30 pzem(PZEM_DEFAULT_ADDR);`

Because I can't:

" PZEMHardSerial:9:35: error: invalid conversion from 'int' to 'HardwareSerial*' [-fpermissive]
 PZEM004Tv30 pzem(PZEM_DEFAULT_ADDR);
                                  ^
In file included from C:\Users\CD86~1\AppData\Local\Temp\arduino_modified_sketch_27982\PZEMHardSerial.ino:1:0:
D:\���������\Arduino\libraries\PZEM-004T-v30-master/PZEM004Tv30.h:65:5: note:   initializing argument 1 of 'PZEM004Tv30::PZEM004Tv30(HardwareSerial*, uint8_t)'
     PZEM004Tv30(HardwareSerial* port, uint8_t addr=PZEM_DEFAULT_ADDR);
     ^~~~~~~~~~~
exit status 1
invalid conversion from 'int' to 'HardwareSerial*' [-fpermissive]
"

MikeHales1, 2021-03-03:
Hi,

I'm getting similar problems trying to use the library on an Arduino UNO Wifi Rev 2 board. Errors reported are:
O:\Documents\Arduino\libraries\PZEM004Tv30\PZEM004Tv30.cpp:54:23: error: conflicting declaration 'arduino::HardwareSerial Serial'
 extern HardwareSerial Serial;
                       ^~~~~~
In file included from C:\Users\Mike\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/Arduino.h:140:0,
                 from O:\Documents\Arduino\libraries\PZEM004Tv30\PZEM004Tv30.h:41,
                 from O:\Documents\Arduino\libraries\PZEM004Tv30\PZEM004Tv30.cpp:23:
C:\Users\Mike\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/UART.h:182:20: note: previous declaration as 'UartClass Serial'
   extern UartClass Serial;
                    ^~~~~~
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.

As the error suggests, it is something to do with the line 'extern HardwareSerial Serial;' but I haven't tracked down what. I will try the comments above.

Just as a further note, I have used the library on the Arduino MEGA without any problems, so I don't understand why the UNO Wifi is causing problems. The only difference is the UNO uses Serial1 not Serial3; I will go back and see if it works on the MEGA on Serial1, although I doubt this is the cause due to the code line the error is reporting.

Any help on fixing this would be great.

Thanks.

MikeHales1, 2021-03-03:
As I suspected, it runs fine on Arduino MEGA on Serial1. 
Might need to delve a bit deeper into why the MEGA and UNO Wifi compile differently as they both use Arduino.h which in turn uses HardSerial.h.
Interestingly the MRK 1010 is also a Wifi board.

zygisjas, 2021-03-16:
> Serial1

Could you please clarify your changes? I have same problem on Arduino Nano 33 IoT

MikeHales1, 2021-03-19:
Hi,

I tried to work out what was wrong with the library but gave up.

I'm currently writing a function to read the values without using the
library. In essence you simply send a number of bytes and then interpret
the bytes received. Not too difficult.

Regards,

Mike

zygisjas, 2021-03-20:
> Hi, I tried to work out what was wrong with the library but gave up. I'm currently writing a function to read the values without using the library. In essence you simply send a number of bytes and then interpret the bytes received. Not too difficult. Regards, Mike
> […](#)

What board do you use?

mandulaj, 2021-07-16:
I will close this as stale. In case you still face problems even with the new version, feel free to reopen the issue.

closed 2021-07-16 by mandulaj

MikeHales1, 2021-10-12:
This is a shortened file I used. It might not compile. However, it shows how to initiate Serial1 to use with 3 PZEM modules on the same serial port. The routine to read the PZEMs is setup to use addresses 10, 11 and 12. To work out the correct codes to send I used the CRC calculator in the PZEM library, as they are constants you only need to do this once. Then it is really simple to read the modules. As I only pole them at a maximum rate of every 2 seconds, I don't need to check if new data is available. Good luck.

#include <TimeLib.h>
#include <SD.h>

/*  PZEM variables,  constants, etc
 *  
 *  Floating point arrays are used to store the data from the PZEM modules:
 * The array is arranged:
 *  //[0] - Volts    80 - 260V
 *  //[1] - Amps     0 - 100A
 *  //[2] - Watts   0 - 23kW        May need to check this as format changes > 1000W
 *  //[3] - Wh      0 - 9999.99kWh  May need to chack this as changes from Wh to kWh > 9999Whr
 *  //[4] - Hz      45 - 65Hz
 *  //[5] - pf      0 - 1
 *  //[6] - Alarm
 */
  float PZEM10H [7];   //Household values
  float PZEM11S [7];   //Solar panel values
  float PZEM12I [7];   //Immersion heater values
  int today;           //When the day of the month changes it is midnight.
  #define House 10     //PZEM address for House
  #define Solar 11     //PZEM address for solar panels
  #define Immer 12     //PZEM address for immersion heater

  int SDFreq = 10;      //Number of seconds between save to SD card
                        //Make sure is a simple divisor of 60 eg 60, 30, 20, 10, 5, 3, 2, 1
/* UdpNtpClient variables
 *  
 *  Read the time for the web server
 */

  int CurrentSecond;  //Store current second to check if it changes
  int CurrentHour;    //Store current hour to check if if changes
  int CurrentMonth;   //Store current month to check if it changes
  int CurrentDay;     //Store current day to check when it changes
  char MonthFile[14];     //Filename for storing monthly PZEM data
  char MonthFileW[14];    //Filename for storing hourly Wh readings
  char YearFile[14];      //Filename for storing annual data

void setup() {

  //PZEM comms is 9600 baud, 8 bits, 1 stop bit, no parity
  Serial1.begin (9600, SERIAL_8N1);

  CurrentSecond = second();   //Initialise the current second
  FileNames (MonthFile, MonthFileW, YearFile); //Initialise the monthly and annual filename

}

void loop() {

  //Routines to run every second
  if (CurrentSecond != second()) {
    //Read the PZEMs every second
    if ((second() % SDFreq) == 0)  {
    ReadPZEM (PZEM10H, House);            //Read the house values
    ReadPZEM (PZEM11S, Solar);            //Read the solar values
    ReadPZEM (PZEM12I, Immer);            //Read the immersion heater values
    }
   CurrentSecond = second();
  }
}

//Start of subroutines

void PZEMtoSD (float PZEMData[7], char Line[40]) {
char Part[10];
  //Format float array into char array for output to SD card
  dtostrf (PZEMData[0],5,1, Part);  //Convert volts
[... 73 lines cut ...]
                   Buff12[10] |
                   Buff12[11] << 24 |
                   Buff12[12] <<16) / 10.0;
    ReadData[3] = (Buff12[13] << 8 | //Raw energy in 1Wh
                   Buff12[14] |
                   Buff12[15] << 24 |
                   Buff12[16] << 16) / 1000.0;
    ReadData[4] = (Buff12[17] << 8 | // Raw frequency in 0.1Hz
                   Buff12[18]) / 10.0;
    ReadData[5] = (Buff12[19] << 8 | // Raw pf in 0.01
                   Buff12[20] ) / 100.0;
    ReadData[6] = (Buff12[21] << 8 | // Raw alarm value
                   Buff12[22]);
  }
  else {
    //If error then set to NAN - 'Not a Number'
    ReadData[0] = NAN;
    ReadData[1] = NAN;
    ReadData[2] = NAN;
    ReadData[3] = NAN;
    ReadData[4] = NAN;
    ReadData[5] = NAN;
    ReadData[6] = NAN;
  }
}