mandulaj/PZEM-004T-v30 #40: How to enable multiple slave in an same Serial Interface ?

hmaln-kevin, 2020-11-18:
Hi, I'd like do understand how can I use more than one PZEM using the same Serial Interface as described in README. I've tested some ways but none worked.
PS: I'm starting to use Github these days, sorry if it isn't the correct space. Thanks!

sergiocntr, 2020-11-18:
Hi hmain-kevin, ok to ask it is not a problema,try to look inside the closed issues ,there are some about this question :)

hmaln-kevin, 2020-11-23:
Hi, again! I've been trying to connect 3 PZEM-004T to my ESP32. I started reading the closed Issues and found some great help in this way. I set the adress of each PZEM and connect all togheter (ESP TX in all PZEMs RX and ESP RX in all PZEMs TX) in the same serial controller. For instance, I'm using Serial2 which for ESP32 are pins 16 and 17. However, I got problems. To be brief, when I start the system I got errors in all measures, at the moment when I disconnect one of the PZEMs and connect again all measures started to be properly shown. The same happen if I disconnect and connect the grid. Bellow is the code that I'm using.
I'm grateful for any help.

#include <PZEM004Tv30.h>

PZEM004Tv30 pzem1(&Serial2, 0x12);
PZEM004Tv30 pzem2(&Serial2, 0x13);
PZEM004Tv30 pzem3(&Serial2, 0x42);

void setup() {
 pinMode(2,OUTPUT);
 Serial.begin(115200);
}

void loop() {

   Serial.println("PZEM1=====================");
   float voltage = pzem1.voltage();
   if(!isnan(voltage)){
       Serial.print("Voltage: "); Serial.print(voltage); Serial.println("V");
       digitalWrite(2,HIGH); //just for debug when Serial monitor isn't connected
   } else {
       Serial.println("Error reading voltage");
       digitalWrite(2,LOW);
   }

   float current = pzem1.current();
   if(!isnan(current)){
       Serial.print("Current: "); Serial.print(current); Serial.println("A");
   } else {
       Serial.println("Error reading current");
   }

   float power = pzem1.power();
   if(!isnan(power)){
       Serial.print("Power: "); Serial.print(power); Serial.println("W");
   } else {
       Serial.println("Error reading power");
   }

   float energy = pzem1.energy();
   if(!isnan(energy)){
       Serial.print("Energy: "); Serial.print(energy,3); Serial.println("kWh");
   } else {
       Serial.println("Error reading energy");
   }

   float frequency = pzem1.frequency();
   if(!isnan(frequency)){
       Serial.print("Frequency: "); Serial.print(frequency, 1); Serial.println("Hz");
   } else {
       Serial.println("Error reading frequency");
   }

   float pf = pzem1.pf();
   if(!isnan(pf)){
       Serial.print("PF: "); Serial.println(pf);
   } else {
       Serial.println("Error reading power factor");
   }

   Serial.println();
   Serial.println("PZEM2=====================");

   voltage = pzem2.voltage();
   if(!isnan(voltage)){
       Serial.print("Voltage: "); Serial.print(voltage); Serial.println("V");
   } else {
       Serial.println("Error reading voltage");
   }

   current = pzem2.current();
   if(!isnan(current)){
       Serial.print("Current: "); Serial.print(current); Serial.println("A");
   } else {
       Serial.println("Error reading current");
   }

   power = pzem2.power();
   if(!isnan(power)){
       Serial.print("Power: "); Serial.print(power); Serial.println("W");
   } else {
       Serial.println("Error reading power");
   }

   energy = pzem2.energy();
   if(!isnan(energy)){
       Serial.print("Energy: "); Serial.print(energy,3); Serial.println("kWh");
   } else {
       Serial.println("Error reading energy");
   }

   frequency = pzem2.frequency();
   if(!isnan(frequency)){
       Serial.print("Frequency: "); Serial.print(frequency, 1); Serial.println("Hz");
   } else {
       Serial.println("Error reading frequency");
   }

   pf = pzem2.pf();
   if(!isnan(pf)){
       Serial.print("PF: "); Serial.println(pf);
   } else {
       Serial.println("Error reading power factor");
   }

   Serial.println();
   Serial.println("PZEM3=====================");

   voltage = pzem3.voltage();
   if(!isnan(voltage)){
       Serial.print("Voltage: "); Serial.print(voltage); Serial.println("V");
   } else {
       Serial.println("Error reading voltage");
   }

   current = pzem3.current();
   if(!isnan(current)){
       Serial.print("Current: "); Serial.print(current); Serial.println("A");
   } else {
       Serial.println("Error reading current");
   }

   power = pzem3.power();
   if(!isnan(power)){
       Serial.print("Power: "); Serial.print(power); Serial.println("W");
   } else {
       Serial.println("Error reading power");
   }

   energy = pzem3.energy();
   if(!isnan(energy)){
       Serial.print("Energy: "); Serial.print(energy,3); Serial.println("kWh");
   } else {
       Serial.println("Error reading energy");
   }

   frequency = pzem3.frequency();
   if(!isnan(frequency)){
       Serial.print("Frequency: "); Serial.print(frequency, 1); Serial.println("Hz");
   } else {
       Serial.println("Error reading frequency");
   }

   pf = pzem3.pf();
   if(!isnan(pf)){
       Serial.print("PF: "); Serial.println(pf);
   } else {
       Serial.println("Error reading power factor");
   }

   Serial.println();
   delay(2000);
}

mandulaj, 2020-11-23:
Incorrect as in you get some random values or do you see the "Error reading ..." message? Strange that after you power cycle, it starts working again....

hmaln-kevin, 2020-11-23:
I see the "Error reading.." message. It's really strange! I tried change the PZEMs, change the ESP32, rechange the adress but it didn't work. When I use only two PZEMs it works fine but at the moment that I put the third it starts to show the "Error reading.." message, until I disconnect and connected one of these, as I said.

hmaln-kevin, 2020-11-23:
Finally I got it! I solved changing the serial port to 25RX and 26TX through the commands:

#include <PZEM004Tv30.h>
#include <HardwareSerial.h>

HardwareSerial MeterSerial(2);
PZEM004Tv30 pzem1(&MeterSerial, 0x12);
PZEM004Tv30 pzem2(&MeterSerial, 0x13);
PZEM004Tv30 pzem3(&MeterSerial, 0x11);

void setup() {
 Serial.begin(115200);
 MeterSerial.begin(9600,SERIAL_8N1,25,26);
}
.
.
.
Thanks for your reply and for your work @mandulaj

closed 2020-11-23 by hmaln-kevin

mandulaj, 2020-11-23:
Hmm that is interesting, I am still curious as to the real reason behind the issue.