mandulaj/PZEM-004T-v30 #41: Wemos D1 Mini / Software Serial compile error

Jex2512, 2020-11-22:
Hello,

First, thanks U for this library ! 👍 

I want to use her with a "Wemos D1 Mini" board. When i compile the exemple program named "PZEMSoftwareSerial" i have the following error code :
`C:\Users\ADN\Documents\Arduino\libraries\PZEM-004T-v30-master\PZEM004Tv30.cpp:456:42: error: 'class SoftwareSerial' has no member named 'listen'`.

I have been looking for a solution on internet for several days but without success...

For additionnal information, this program work very well on a "Arduino Uno" board.

Thanks U in advance

mandulaj, 2020-11-23:
There seems to be an issue with the Software serial on the ESP boards. Could you try disabling the software serial by placing this line before the library import:
```c
#define PZEM004_NO_SWSERIAL

#include <PZEM004Tv30.h>
```

Then  when you create the PZEM object, pass it the hardware serial interface you want to use: 
```c
PZEM004Tv30 pzem(&Serial);
```
You can then connect your PZEM RX TX to pins TX RX respectively (TXD0 and RXD0).

Unfortunately, you won't be able to use the Serial for printing debug information when you do it this way... Ideally you would use `Serial1` instead of `Serial` which the ESP8266 does have! But for some reason, the Wemos D1 Mini only has the TX pin for Serial1 provided on pin D4. We also need the RX pin unfortunately for the PZEMs to work.

Also I noticed on the pinout, there is a RXD2 and TXD2, but they are only an alternative pinout for the default `Serial` You can switch between the *XD0 and *XD2 by calling `Serial.swap()`
I am not sure how this is connected internally, but if the debugger isn't connected to the  *XD2, you could try switching back and forth between the two pins depending if you need to write stuff to your debug terminal or if you want to communicated with the PZEMs. Something like this:

```c
#define PZEM004_NO_SWSERIAL // Disable the software serial

#include <PZEM004Tv30.h>

// Use hardware Serial(0) but on pins D8 and D7
PZEM004Tv30 pzem(&Serial);

void setup() {
  // Serial.begin(9600); // No need to do this, called inside the init function of PZEM, but 
}

void loop() {
    Serial.swap(); // Swap to the *XD2 pins for talking to PZEM
    float voltage = pzem.voltage();

    Serial.swap(); // Swap back to *XD0 pins for printing debug info....
    if(!isnan(voltage)){
         Serial.print("Voltage: "); Serial.print(voltage); Serial.println("V");
    } else {
        Serial.println("Error reading voltage");
    }

    Serial.swap(); // Swap to the *XD2 pins again... 
    float current = pzem.current();
    Serial.swap(); // Swap back to *XD0 pins etc...

    if(!isnan(current)){
        Serial.print("Current: "); Serial.print(current); Serial.println("A");
    } else {
        Serial.println("Error reading current");
    }

    Serial.println();
    delay(2000);
}

```
I have no idea if this would work, but it is a thought I had you could try. Unfortunately, I don't have a Wemos D1 so can not test it personally. I would be glad if you could let me know how this works.

Jex2512, 2020-11-25:
Thanks U for this reponse !

These two days i've test use of Serial.swap but without success...

I've upgrade my ESP8266 library and now i can compile "PZEMSoftwareSerial" example program without error. But after upload program on the Wemos board i've exception error "exception (28)" periodically.

So i think i will use hardware serial (it works very well), and for debug information (if i need) i will use an SD card to write text.

Have a nice day

mandulaj, 2021-07-16:
I will close this issue as it should be now fixed. Feel free to reopen it again, if you come across the same problem with the new version of the library.

closed 2021-07-16 by mandulaj