mandulaj/PZEM-004T-v30 #44: Measuring speed

zered4, 2021-01-21:
Hello,

dont know if its an issue but is there some limitation to measuring only once a second? And if so is there a way to make the process faster. It looks like the measuring is happening faster but data get "updated" only once a second. What should i do if i want to measure faster?

 I alogizige, english is not my first language and Thank you all for your answers.

mandulaj, 2021-01-23:
Hi @zered4, thank you for the question. This issue has been discussed before in #37. In summary, by default the measurements should be updated (by communicating with the PZEM) every 200ms, (Controlled by the `#define UPDATE_TIME 200` in `PZEM004Tv30.cpp`) However we are actually limited by the internal sampling speed of the PZEMs which is around 1s. So unfortunately, there is not much that can be done... I will close this issue as a duplicate.

mandulaj, 2021-01-23:
Duplicate of #37

closed 2021-01-23 by mandulaj