mandulaj/PZEM-004T-v30 #47: pzem-004t v3 reset energy alone after => 60kw

capblack, 2021-01-28:
hello I built an energy meter of 3 faces with pzem-0004T v3 and I have the problem that the faces reset the accumulated energy after the 60kw approximately.
is a bug? thanks

https://youtu.be/ekmTViQi_cU

mandulaj, 2021-01-29:
Hey @capblack thats very odd, the specs range should be 0-9999.99kWh according to the datasheet, but truth be told, I could not test it as I don't have such an energy consumption. The video though is private so I can't see it. Would you mind setting it as unlisted instead?

capblack, 2021-01-29:
OK thanks for you reply and, in the video you can see the unit , but not the reset in question, that is my average consumption per two months about 1200kw , but @ the start a say is very precise, after 100kw the star to show 30% less and after feew weeks 70% less , and i say there is a problem , and I see the ridings come to 0 after aprox pass 60 kw
[image]

mandulaj, 2021-01-29:
Hmm that is definitely a strange behavior... Sounds like some sort of a overflow. Could be in the PZEM or in the library code. I can't quite see anything obvious in the library code at the moment but I don't want to straight out blame it on the chip. I will have a closer look at this issue once I get around testing the library next week.

Unfortunately, I don't have power consumption quite high enough to test this thoroughly. I will try simulating it somehow but it would be really useful if you could send me the raw numbers that are coming from the PZEM chips. 

Currently there is no debug mode (will add that to the TODO list). Would you be so kind and modify the library slightly in order to print out the raw response during the value update?

If you could add this code:
```c++
Serial.println("Raw response:");
for(int i = 0; i < 25; i++){
    Serial.println(response[i], HEX);
}
Serial.println("End");
```

 after line 379 in the `PZEM004Tv30.cpp` library: https://github.com/mandulaj/PZEM-004T-v30/blob/e7f220220e5f765f5e1fd9963b9d69a053296c0e/PZEM004Tv30.cpp#L377-L379 and send me the serial output for your devices with Energy over 60kWh.

Note that this will print the raw response from the PZEMs to the standard `Serial` device. Change it to what ever serial interface you use for debugging (like for example `Serial2`).

## Temporary solution
In case you are reporting the energy to some sort of a database, you can set the code to reset the internal  energy counter of the PZEMs with the `.resetEnergy()` method once it gets over a certain margin (60kWh). In the database, you can keep track, how many times you had to reset and from that calculate the true aggregate energy. It's not a perfect solution but in case it really is the limitation of the chip, it will be the only option. 

I have had in mind for a while to perform a rigorous analysis of the true capabilities of the module and compare them to the manufacturer specifications. I would not be too surprised if the datasheet claims are slightly too optimistic.

morganflint, 2021-02-28:
Maybe the OP is having a similar problem that was addressed [here](https://community.blynk.cc/t/pzem-004t-v3-0-and-nodemcu-wemos-mini-running-on-blynk-how-to-procedure/39338/61).
Basically, it was due to using only 2 of the 4 bytes of the energy registers, so it resets when it reaches 65532 Wh (65.5kWh).
That was with another library (ModbusMaster), I haven't analyzed if this one has the same problem, but it's suspicious the OP has the problem around the same "magic number"

capblack, 2021-02-28:
Hello yes i change the busmaster with standart library and al is ok now no
reset energy ,and can see the acumulated in history thanks for

brunosso, 2022-03-07:
I don't understand if this issue is fixed or not.
Thanks

pioioTwo, 2022-03-22:
ia have problem, my code idem with examples

pioioTwo, 2022-03-22:
my pzem its reset energy after 65 Wh

capblack, 2022-03-22:
yes i fix the problem , was the totals storage variable had an unsing_intiger, when I changed it to a long fix the bug :)

pioioTwo, 2022-03-23:
can you give me an example

mandulaj, 2022-03-24:
Yes @capblack, could you push your changes as a pull request. I am quite curious what the issue is. I had a look over the code and I am using uint32_t which should be large enough to hold the energy values. 

If possible, I would like to avoid going to a long but perhaps its the unsigned nature that's causing the problem. Would have to investigate closer.

capblack, 2022-03-24:
Hi Jakub You coul download full code herehttps://drive.google.com/drive/folders/1W3jGJp451KJDXy7BvQeOkArqFwXoovCk?usp=sharing and you could check it on https://www.youtube.com/watch?v=ekmTViQi_cU be in touch Sent from Mail for Windows From: Jakub MandulaSent: Thursday, March 24, 2022 1:21 PMTo: mandulaj/PZEM-004T-v30Cc: capblack; MentionSubject: Re: [mandulaj/PZEM-004T-v30] pzem-004t v3 reset energy alone after => 60kw (#47) Yes @capblack, could you push your changes as a pull request. I am quite curious what the issue is. I had a look over the code and I am using uint32_t which should be large enough to hold the energy values.If possible, I would like to avoid going to a long but perhaps its the unsigned nature that's causing the problem. Would have to investigate closer.

123salee, 2022-11-14:
TTGO T-CALL ESP32 CODE??

mandulaj, 2022-11-14:
@123salee could you please be a bit more specific? Is this related to this issue? Do you want example code? Do you have a problem?

123salee, 2022-11-15:
Good day, do you have an example code for pzem004t using ttgo tcall esp32?
I have tried your code for hardserial but i think it is not for ttgo.thanks

capblack, 2022-11-15:
hello not only for esp8266, but I think it should not be very difficult to adapt it sorry Sent from Mail for Windows From: 123saleeSent: Tuesday, November 15, 2022 2:05 AMTo: mandulaj/PZEM-004T-v30Cc: capblack; MentionSubject: Re: [mandulaj/PZEM-004T-v30] pzem-004t v3 reset energy alone after => 60kw (#47) Good day, do you have an example code for pzem004t using ttgo tcall esp32?I have tried your code for hardserial but i think it is not for ttgo.thanks

mandulaj, 2023-02-02:
Closing as resolved.

closed 2023-02-02 by mandulaj