mandulaj/PZEM-004T-v30 #53: Compatibility with Arduino nano 33 IoT

zygisjas, 2021-03-16:
I was trying to connect PZEM004 to Arduino nano 33 IoT board, and this library is using SofwareSerial.h,
I tried to dig deeper, and Nano 33 IoT does not use SoftwareSerial anymore, so I'm not sure how to make
PZEM004 compatible with Nano 33 IoT board, could not find anything in the internet about this isse. 
Maybe someone has an idea how to solve this?

sergiocntr, 2021-03-16:
> internet about this issue.

Ciao Jasiunas, I'm sorry I don't own a board like yours, seems that it
desn't need software serial because you can use different UART interfaces,
much better explained here :
https://stackoverflow.com/questions/57175348/softwareserial-for-arduino-nano-33-iot/57541921#57541921

Let us know your progress.
Sergio.

zygisjas, 2021-03-16:
It was complaining about the Usb Serial conflict in the CPP file, I commented line 54 in CPP.

```
#define UPDATE_TIME     200

#define RESPONSE_SIZE 32
#define READ_TIMEOUT 100

#define PZEM_BAUD_RATE 9600

**//extern HardwareSerial Serial;**

#define DEBUG

// Debugging function;
void printBuf(uint8_t* buffer, uint16_t len){
```

And passing Serial to PZE by using this line:

`PZEM004Tv30 pzem(&Serial1);`

Now code compiles with no problem, but... when I install this code to Arduino nano 33 IoT, the board simply crashes, and I need to 
do hard reset. Still no luck.

zygisjas, 2021-03-17:
Update:

mandulaj, 2021-07-16:
Unfortunately I don't have a Nano 33 IoT. Could you try with the new version v1.1.1 of the library to see if you still face the same issues.

robertoperuzzo, 2022-12-12:
I think this library https://github.com/zygisjas/PZEM-004T-V30-SAMD21 may help you @zygisjas. I'm using it with Arduino MKR 1310 and it works.

robertoperuzzo, 2022-12-12:
For anyone using the same processor's family (Cortex-M0 32-bit SAMD21) the library https://github.com/zygisjas/PZEM-004T-V30-SAMD21 works. I tested with [MKR WAN 1310](https://docs.arduino.cc/hardware/mkr-wan-1310).

Thanks to @zygisjas