mandulaj/PZEM-004T-v30 #54: Read 2 pzem04 with wemos d1 with 2 serial

sanluca, 2021-04-10:
Good morning,

I have a problem with wemos d1 and 2 pzem04T

When wemos d1 turns on it does not read the pzem04, 
I must first connect 1 pzem and after the other pzem, then it starts to read them

the code is attached

thank you very much

`//  

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <PubSubClient.h>
#include <PZEM004Tv30.h>

//PZEM004Tv30 pzem(&Serial3);
PZEM004Tv30 pzem (D5, D6);
PZEM004Tv30 pzem2 (D7, D8);
                    
const char* ssid = "+++++++";
const char* password = "++++++++";
const char* mqtt_server = "192.168.1.1";
long previousMillis = 0;
long interval = 60000;

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

const char* inTopic = "home/enel";
const char* outTopic = "home/enel";
const char* outTopic_enel_gen_v = "home/enel/generale/v";
const char* outTopic_enel_gen_a = "home/enel/generale/a";
const char* outTopic_enel_gen_w = "home/enel/generale/w";
const char* outTopic_enel_gen_kwh = "home/enel/generale/kwh";
const char* outTopic_enel_gen_hz = "home/enel/generale/hz";
const char* outTopic_enel_gen_pf = "home/enel/generale/pf";
const char* outTopic_enel_ap_v = "home/enel/appartamento/v";
const char* outTopic_enel_ap_a = "home/enel/appartamento/a";
const char* outTopic_enel_ap_w = "home/enel/appartamento/w";
const char* outTopic_enel_ap_kwh = "home/enel/appartamento/kwh";
const char* outTopic_enel_ap_hz = "home/enel/appartamento/hz";
const char* outTopic_enel_ap_pf = "home/enel/appartamento/pf";
const char* outIP = "home/enel/ip";

void setup_wifi() {

  delay(5);
  WiFi.begin(ssid);

  while (WiFi.status() != WL_CONNECTED) {
    for(int i = 0; i<500; i++){
      delay(1);
    }
    Serial.print(".");
  }
}

void callback(char* topic, byte* payload, unsigned int length) {
  if ((char)payload[0] == '1') {
    pzem.resetEnergy();
    pzem2.resetEnergy();
  
  }
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("Enel")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish(outTopic, "enel booted");
      client.subscribe(inTopic);
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      for(int i = 0; i<5000; i++){
        delay(1);
      }
    }
  }
}

void send_ip(){
  unsigned long currentMillis = millis();
  if(currentMillis - previousMillis > interval) {
    client.publish(outIP, WiFi.localIP().toString().c_str());
  }
}

void read_enel_gen(){

  float volt = pzem.voltage();
  Serial.print("Voltage: ");
  Serial.print(volt);
  Serial.println("V");

  float cur = pzem.current();
  Serial.print("Current: ");
  Serial.print(cur);
  Serial.println("A");

  float powe = pzem.power();
  Serial.print("Power: ");
  Serial.print(powe);
  Serial.println("W");

  float ener = pzem.energy();
  Serial.print("Energy: ");
  Serial.print(ener,3);
  Serial.println("kWh");

  float freq = pzem.frequency();
  Serial.print("Frequency: ");
  Serial.print(freq);
  Serial.println("Hz");

  float pf = pzem.pf();
  Serial.print("PF: ");
  Serial.println(pf);

  char buffer[10];
  dtostrf(volt,2,2, buffer);
  client.publish(outTopic_enel_gen_v,buffer);
  dtostrf(cur,2,2, buffer);
  client.publish(outTopic_enel_gen_a,buffer);
  dtostrf(powe,2,2, buffer);
  client.publish(outTopic_enel_gen_w,buffer);
  dtostrf(ener,2,2, buffer);
  client.publish(outTopic_enel_gen_kwh,buffer);
  dtostrf(freq,2,2, buffer);
  client.publish(outTopic_enel_gen_hz,buffer);
  dtostrf(pf,2,2, buffer);
  client.publish(outTopic_enel_gen_pf,buffer);

  }

  void read_enel_ap(){

  float volt = pzem2.voltage();
  Serial.print("Voltage2: ");
  Serial.print(volt);
  Serial.println("V");

  float cur = pzem2.current();
  Serial.print("Current: ");
  Serial.print(cur);
  Serial.println("A");

  float powe = pzem2.power();
  Serial.print("Power: ");
  Serial.print(powe);
  Serial.println("W");

  float ener = pzem2.energy();
  Serial.print("Energy: ");
  Serial.print(ener,3);
  Serial.println("kWh");

  float freq = pzem2.frequency();
  Serial.print("Frequency: ");
  Serial.print(freq);
  Serial.println("Hz");

  float pf = pzem2.pf();
  Serial.print("PF: ");
  Serial.println(pf);

  char buffer[10];
  dtostrf(volt,2,2, buffer);
  client.publish(outTopic_enel_ap_v,buffer);
  dtostrf(cur,2,2, buffer);
  client.publish(outTopic_enel_ap_a,buffer);
  dtostrf(powe,2,2, buffer);
  client.publish(outTopic_enel_ap_w,buffer);
  dtostrf(ener,2,2, buffer);
  client.publish(outTopic_enel_ap_kwh,buffer);
  dtostrf(freq,2,2, buffer);
  client.publish(outTopic_enel_ap_hz,buffer);
  dtostrf(pf,2,2, buffer);
  client.publish(outTopic_enel_ap_pf,buffer);

  }

 

void setup() {
  
  Serial.begin(115200);
  setup_wifi();                   // Connect to wifi 
  client.setServer(mqtt_server, 1883);
 // client.setCallback(callback);

  Serial.print("Reset Energy");
//  pzem.resetEnergy();
//  pzem2.resetEnergy();

 // Serial.print("Set address to 0x42");
//  pzem.setAddress(0x42);

  // Hostname defaults to esp8266-[ChipID]
   ArduinoOTA.setHostname("enel");
  // START OTA
  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_FS
      type = "filesystem";
    }

    // NOTE: if updating FS this would be the place to unmount FS using FS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
//END OTA
}

void loop() {
  if (!client.connected()) {
    reconnect();
  }
  ArduinoOTA.handle();
  client.loop();
 read_enel_gen();
 delay(2000);
 read_enel_ap();
 delay(2000);
 send_ip();
}`

sergiocntr, 2021-04-11:
Ciao Luca

you cannot use the D8 as an input pin because boot fail if pulled HIGH.
Try as follow:
First change the PZEM address, set one 0X42 and the other 0X43 ,you have to connect one at a time and set the address individually,use  D5 and D6 pin.
In your sketch yuo commented that lines. Set the device address and read it to be sure the address is OK.
Once you have two addresses change constructor :
`PZEM004Tv30 pzem (D5, D6, 0X42);
 PZEM004Tv30 pzem2 (D5, D6, 0X43);
`

sanluca, 2021-04-11:
Ciao Sergio,

Now works

thank you very much

mandulaj, 2021-07-16:
Closing as resolved.

closed 2021-07-16 by mandulaj