mandulaj/PZEM-004T-v30 #58: Very often errors in response

rogerknop, 2021-04-26:
Hi,
I am using PCEM-004T-v30 with a NodeMCU on PINs D5 and D6.
It looks almost good, but I get a lot of errors that data could not be read.
I checked your code and it happens in updateValues because "recieve(response, 25) != 25" fails.
When comment that line out it works.
I am only interested in Power and probably Power is updated correctly even if this check fails.
Any idea?
Thanks & regards, 
Roger

ASMotionLab, 2021-07-15:
I'm not sure about the receive response 25 thing, but a more general question - are you using an esp32 or esp8266 NodeMCU?
If it's an esp32 version and you're using SoftwareSerial, it's very unreliable - I analysed the comms using a logic analyser and they were full of frame errors. Make sure it's using HardwareSerial and remap the pins manually if you have to.

rogerknop, 2021-07-16:
I am using NodeMCU.
But I will check that.
Thanks! Roger

mandulaj, 2021-07-16:
As @ASMotionLab said, SoftwareSerial for ESP32 is unreliable. It will be discontinued for that platform in the future versions until we can find a fix.

mandulaj, 2021-07-16:
I will close this issue now. If you have any further questions or comments regarding SoftwareSerial on the ESP32, please put them in the issue #70

closed 2021-07-16 by mandulaj