mandulaj/PZEM-004T-v30 #59: ESP8266 - Multiple PZEM004Tv3 address assigned devices on same pins - Software Serial

ejbrown1, 2021-05-31:
HI, Initially, many thanks to OLEH and MANDULAJ for their efforts.
I am new to this site, and as I have setup, and painfully tested over the last couple of days, various methods of connecting and utlizing the library. (And I did some debugging lines in the library to learn it a bit better.)  I'm not sure if what I will speak about is due to an unintentional bug, or in the end, what I will explain below, opened up a pretty nice option for me.

For me, I wanted to use multiple addressed PZEM devies on the same two sets of pins, as intend to have at least 7 of them connected to one board.  Setting the address up using the specific sketch for this requirement was easy, but initially I did not realize you do need the AC 120/240 Supply connected, for this to be successful.  Note:, if you do this after the sketch is running, you can hit the reset, on the ESP8266 board to rerun the "setAddress" command.

Once all the PZEM boards, are setup with unique Addresses, I started working on my project to read them all.
I didn't start small, as I added in all my Gmail, Blynk, NTPTime, etc code and wrapped this code up, as If I knew from the discussions how this board worked.  I also appreciated the request by users wanting to use arrays for loops, and simplified coding.  I tried many examples, using the array/For loop possibilities, but was having issues with the returned data.  I had to scale bakc to a basic sketch to try and figure out what the results were not as expected, such as like one user on this site, having only the last PZEM instance return data results when only the first address was setup yet.  I did not see the answer to that user, but that Issue was closed.
I was having similar issues.
In the end I found a solution that fixed what that user experienced, as well as what other users wanted, as in an array, or 
more simply, a simple for loop to read all PZEM's in some simple fashion. 
I suspect my issues were in the fact I was trying to use seperate instances of the PZEM library on the same pins, with a different address, defined as follows:
`PZEM004Tv30  pzem1(D5, D6, addr1);
PZEM004Tv30  pzem2(D5, D6, addr2);
PZEM004Tv30  pzem3(D5, D6, addr3);`
With only one PZEM initially connected with address 1.
The problem from this, was my reading pzem1 failed, and my reading pzem3 gave the results of addr1.  Not consistent with expectations.

I also tried created instances of the PZEM like this:
`PZEM004Tv30 pzemObj[] = {PZEM004Tv30(D5, D6, 0x01), PZEM004Tv30(D5, D6, 0x02), PZEM004Tv30(D5, D6, 0x03), PZEM004Tv30(D5, D6, 0x04), PZEM004Tv30(D5, D6, 0x05), PZEM004Tv30(D5, D6, 0x06), PZEM004Tv30(D5, D6, 0x07)};
`
AND Like this
`PZEM004Tv30 pzemObj[] = {PZEM004Tv30 pzem1(D5, D6, 0x01), PZEM004Tv30 pzem2(D5, D6, 0x02), PZEM004Tv30 pzem3(D5, D6, 0x03), PZEM004Tv30 pzem4(D5, D6, 0x04), PZEM004Tv30 pzem5(D5, D6, 0x05), PZEM004Tv30 pzem6(D5, D6, 0x06), PZEM004Tv30 pzem7(D5, D6, 0x07)};`

AND some other configurations using the NEW constructor.
What I found did work to match the right address to the object was using the New Construction before reading the device, and also deleting the constructor and setting to NULL as some code i found suggested to do, but as also warned in some posts, this could be a memory issue .  It was, it gave great results for an hour of requests that were done every second, then failed wdt exception 29.

SO i started looking closer at the library code using the newly installed 'Visual Studio Code" software.  For understanding, as well as to insert some Serial.print statements in, so I could see what was happening a bit better, and why my sketch was not relating to the right address/device.  IE. Usually the last defined device received the results of a different board if you recall, so the address field seems to get mixed up in the library instance(s).  I think this possibly is due to the expectation that all defined boards will be connected and powered up. (maybe).  But if you code is setup to read 7 boards, on the same set of pins, and not all boards are available when the library is initialized,....issues?

Anyway, lets get to what worked for me, and worked well so far, but has not been discussed yet.  Maybe i was be scolded for inappropriate use of code rules, as I am not at the level that of understanding that many here are.  But here goes.

Step1.
In the library header, i moved the "init" declaration from the Private, to the Public declartions.

Step 2 Sketch declare your PZEM library instance like this.

`PZEM004Tv30 pzem = PZEM004Tv30(rxPin, txPin, addr);   // This works, just have to call pzem.init(addr) before reading Addressed device, to change address calls`

Do this only once. as in do not do it for your other PZEM devices, even though they have other addresses.  Good new is , this method only requires one instance of the library to read all 247 other possible addressed PZEM devices.

Then to read/display the other devices you can do it very simply like this:
`#include <Arduino.h>
#include <PZEM004Tv30.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
#include <stdio.h>

uint8_t rxPin = 14; //D5  type to match library
uint8_t txPin = 12;  //D6  type to match library
uint8_t addr = 0x0001;
uint8_t addr2 = 0x0002;
uint8_t addr3 = 0x0003;
uint8_t addr4 = 0x0004;
uint8_t addr5 = 0x0005;
uint8_t addrA[7] = {0x0001, 0x0002, 0x0003, 0x0004, 0x0005, 0x0006, 0x0007};

float v[7];
float p[7];
float pm[7];
float hz[7];
float a[7];
float pf[7];

PZEM004Tv30 pzem = PZEM004Tv30(rxPin, txPin, addr);   // This works, just have to call pzem.init(addr) before reading Addressed device, to change address calls

// ************************************************   SETUP       SETUP       SETUP       SETUP       SETUP       SETUP       *********************************************
void setup() {

  Serial.begin(115200);
[... 39 lines cut ...]
      Serial.print(" pm: " + String(pm[x])+"   ");
      Serial.print(" Hz: " + String(hz[x])+"   ");
      Serial.print(" PF: " + String(pf[x])+"   ");
    }//End If Device values available
    
    
  Serial.println(" Address: " + String(pzem.getAddress())+"    ");
  
  } // end for x (Device Addr)
  Serial.println();
  
}
`

So you can see I am changing the address of the pzem instance by calling the init subroutine with the desired address, before reading the values again.  The extra benefit of calling the init subroutine, is that it resets teh last know device read time, so that you can read the device with the new address right away.  Intent of the library was to delay reads of the device, as the PZEM hardware is only supposed to be read every (second?), but we are chnaging the device being read via the address call change.
As you can also seem this simplifies the reading of the device values.  This type of method was requested by one of the users in the posts as well.
So, hopefully I will not recieve a roast for my hack methods, but so far this works well for me.  No memory issues.  simple code structure.   It can handle a board dropping from an available state..

I hope this helps someone else, achieve data access from the PZEM in the same manner as I had wanted.
IE 2 pins to read many addressed PZEM devices in an efficient arrayish manner.  Utilizing an ESP8266 NodeMCU in this case.
And hopefully there is not some reason, such as memory, or bad program structure that this is not an ok method.

So not sure if this is great idea, bad idea, bug, help or new wiki help?  Please be kind.
Thanks!
Earl

vortigont, 2021-06-05:
For me it works pretty fine with different object instances - one per PZEM device. There is no need to init() it every time. Only set individual addresses once.

Something like
`
PZEM004Tv30 pzemb(&PzemSerial, 22, 17, 0xb); // rx 22, tx 17
PZEM004Tv30 pzemc(&PzemSerial, 22, 17, 0xc); // rx 22, tx 17

Serial.print("Voltage b: "); Serial.print(voltage); Serial.println("V");
Serial.print("Voltage c: "); Serial.print(voltage); Serial.println("V");
`

closed 2021-07-16 by mandulaj

deri257, 2022-06-05:
Hello,
Can you share your wiring for 3pzem004t with esp32 ? 

Thanks.

morganflint, 2022-06-06:
I described it [here](https://github.com/mandulaj/PZEM-004T-v30/issues/33#issuecomment-677811222). 

I'm attaching two pictures illustrating it. The second one shows the calculations without a buffer, if I didn't have it wrong back then, up to 5 units can be paralleled without buffer, but I recommend installing it (just a PNP transistor).

I'm also attaching two photos of the setup.
[image]
[image]
[image]
[image]