mandulaj/PZEM-004T-v30 #60: 2 PZEM in parallel on same SoftwareSerial; when 1 loses power the second one stops to work.

Oxmaster, 2021-06-04:
I've setup 2 PZEM-004T-V3 to monitor my small off-grid solar installation, 1st one is on input and 2nd one is on output.

During my testing I just plugged in both to the same socket, so they are both on at the same time. I've got an UPS because of semi frequent blackouts, and if either PZEM lose power, both don't work. I've only noticed that after I've deployed it...

I'm not doing anything fancy. Sampling every 500ms and using EWMA to average data that is published every minute to MQTT.

They are hardwired in parallel to Wemos D1 Mini with modded led resistors.

```
PZEM004Tv30 pzem1(D6, D7, 0x01);
PZEM004Tv30 pzem2(D6, D7, 0x02);
```

Is there any way to make one work when the other has lost power other than putting them on separate bus.? Could I leave TX for both on D7 and separate RX on D6, D5?

mandulaj, 2021-06-04:
That's is strange. Since you are using addressing, there should not be any dependence between them. Remember that the chip gets power from the AC side. The 5V is just for the opto-coupler. Since one measures input and the other output, are you sure that both of them don't loose power?

The only other explanation that I can think of is that when one of the PZEMs looses power, it pulls the BUS line low/high preventing the communication for the other PZEMs. I haven't tested that case but if that is the problem, the easiest solution would be using separate BUSes... 

I would have to investigate to see if there is a solution involving only one BUS. 

The custom TX sharing would be possible but not out of the box. Plus if the issues really is that one of the PZEMs hogs, say the TX line, it would not solve your problem.

Let me get back to you on it if I find something.

Oxmaster, 2021-06-04:
I will be separating the buses, after deploying I was seeing some data loss from poor WiFi connection so I will be doing some extensions to place the ESP somewhere closer to the router. Are there any tests that I could perform to help?

mandulaj, 2021-07-16:
If you have time, could you try the new version of the library and see if by chance the issue isn't resolved. However it seems that the offline PZEM is pulling too hard on the bus which causes the problem so probably your solution with a split bus is the best solution. For now I will close this issue.

closed 2021-07-16 by mandulaj