mandulaj/PZEM-004T-v30 #63: How can I setAddress for my multiple PZEM?

Krunchiee, 2021-07-09:
Hi.... I am a newbie on this.
can u help me tp tell step by step how to configure address for every PZEM on same pin on NodeMCU ESP8266?

ASMotionLab, 2021-07-13:
Firstly, you need to send an address set command to a single PZEM, to give it an address.

You do this by not specifying an address in the constructor (eg PZEM004Tv30 pzem(5,4);) so the library will address any PZEM thats on the bus. So make sure you only have 1 connected.

This example sketch which will set the address of the PZEM to "0x07", number 7 basically. Here the RX&TX are connected to pins 5 & 4 - adjust for yours.
```
#include <PZEM004Tv30.h>

PZEM004Tv30 pzem(5,4);

void setup() {
  Serial.begin(115200);
}

uint8_t addr = 0x07; //THIS IS THE ADDRESS

void loop() {
    pzem.setAddress(addr);
    Serial.print("Current address:");
    Serial.println(pzem.getAddress());
    Serial.println();
    delay(1000);
}
```
After this is done, that particular PZEM will only respond when it's addressed explicitly in the 'constructor'.

You can do the same to a few PZEMS, and they will each take the address you specified.

So then, it'll only reply if it's addressed specifically in the constructor.

```
#include <PZEM004Tv30.h>

PZEM004Tv30 pzem1(5,4,0x07);
PZEM004Tv30 pzem2(5,4,0x08);
PZEM004Tv30 pzem3(5,4,0x09);

void setup() {
  Serial.begin(115200);
}

void loop() {

    Serial.println(pzem1.voltage());
    Serial.println(pzem2.voltage());
    Serial.println(pzem3.voltage());
    Serial.println();
    delay(1000);
}
```

mandulaj, 2021-07-13:
This is a very common question. It has been asked and answered many times already. In #13 #19 #29 #32 and many more.... One day, I will convince myself to writing a guide on how to use everything. I promise :smile:

closed 2021-07-16 by mandulaj

Guudiin, 2022-03-04:
How to reset pzem2 = 0x02 ??
Exemplo: 

PZEM004Tv30 pzem1(D5, D6, 0x01); 
PZEM004Tv30 pzem2(D5, D6, 0x02);

For 0x01 = [0x01, 0x42, 0x80, 0x11]

for 0x02 = ???

help

AwanDami, 2023-03-22:
when I try to upload, I get an error like this: no matching function for call to 'PZEM004Tv30::PZEM004Tv30(int, int)

gidiara, 2024-06-27:
Not working with WeAct studio ESP32 -C3