mandulaj/PZEM-004T-v30 #70: Add SoftwareSerial support for ESP32

mandulaj, 2021-07-16:
The EspSoftwareSerial library does not work on the ESP32. As a result, it is not possible to use SW Serial for this library.
Although the ESP32 HW serial is extremely flexible. It would be nice to also offer Software serial for those who need it.

Unless I get many requests for this feature, I will not look much into it.

sergiocntr, 2022-04-25:
@mandulaj esp32 has 3 hw serials on almost all pins .
For me, no need to implement for this specific project what that I don't find in any others.and are not on subject.

mandulaj, 2022-04-25:
I am very well aware of that. Hence why this is a very low priority task. Though I can think of applications where you are forced to move to software serial. So for the sake of completeness it would be nice to have this working.