mandulaj/PZEM-004T-v30 #72: Seeeduino Xiao

diegomariani, 2021-07-23:
How can use your library in Seeeduino Xiao (SAMD proccessor)?

mandulaj, 2021-07-24:
Hey, Sorry I don't have a Seeeduino so can't test it for you. However besides potential problems with SoftwareSerial, I see no reason why it should not work.

diegomariani, 2021-07-25:
I have problems with the Seeeduino.

If you want you can connect with desktopremoto, and we can create a
suitable class for SeeeduinoXiao as well.

With the SoftwareSerial I get the following error with the following
program:

#include <PZEM004Tv30.h>
#include <SoftwareSerial.h>

SoftwareSerial pzemSWSerial(6, 7);

PZEM004Tv30 pzem;

void setup() {

    SerialUSB.begin(115200);

    pzem = PZEM004Tv30(pzemSWSerial);

    while (!SerialUSB) { delay(100); }

    SerialUSB.println("start");

}

void loop() {
    }

>
C:\Users\maria_dchxf68\Desktop\Measurement_PZEM-004T\Measurement_PZEM-004T.ino:
In function 'void setup()':
>
>
> Measurement_PZEM-004T:17:36: error: no matching function for call to
'PZEM004Tv30::PZEM004Tv30(SoftwareSerial&)'
>
>
>      pzem = PZEM004Tv30(pzemSWSerial);
>
>                                     ^
>
> In file included from
C:\Users\maria_dchxf68\Desktop\Measurement_PZEM-004T\Measurement_PZEM-004T.ino:1:0:
>
>
C:\Users\maria_dchxf68\OneDrive\Documenti\Arduino\libraries\PZEM004Tv30-1.1.2\src/PZEM004Tv30.h:93:5:
note: candidate: PZEM004Tv30::PZEM004Tv30()
>
>      PZEM004Tv30(){};
>
>      ^~~~~~~~~~~
>
>
C:\Users\maria_dchxf68\OneDrive\Documenti\Arduino\libraries\PZEM004Tv30-1.1.2\src/PZEM004Tv30.h:93:5:
note:   candidate expects 0 arguments, 1 provided
>
>
C:\Users\maria_dchxf68\OneDrive\Documenti\Arduino\libraries\PZEM004Tv30-1.1.2\src/PZEM004Tv30.h:90:5:
note: candidate: PZEM004Tv30::PZEM004Tv30(HardwareSerial*, uint8_t)
>
>      PZEM004Tv30(HardwareSerial* port, uint8_t addr=PZEM_DEFAULT_ADDR) :
PZEM004Tv30(*port, addr) {};
>
>      ^~~~~~~~~~~
>
>
C:\Users\maria_dchxf68\OneDrive\Documenti\Arduino\libraries\PZEM004Tv30-1.1.2\src/PZEM004Tv30.h:90:5:
note:   no known conversion for argument 1 from 'SoftwareSerial' to
'HardwareSerial*'
>
>
C:\Users\maria_dchxf68\OneDrive\Documenti\Arduino\libraries\PZEM004Tv30-1.1.2\src/PZEM004Tv30.h:86:5:
note: candidate: PZEM004Tv30::PZEM004Tv30(HardwareSerial&, uint8_t)
>
>      PZEM004Tv30(HardwareSerial& port, uint8_t addr=PZEM_DEFAULT_ADDR);
>
>      ^~~~~~~~~~~
>
>
C:\Users\maria_dchxf68\OneDrive\Documenti\Arduino\libraries\PZEM004Tv30-1.1.2\src/PZEM004Tv30.h:86:5:
note:   no known conversion for argument 1 from 'SoftwareSerial' to
'HardwareSerial&'
>
> In file included from
C:\Users\maria_dchxf68\Desktop\Measurement_PZEM-004T\Measurement_PZEM-004T.ino:1:0:
>
>
C:\Users\maria_dchxf68\OneDrive\Documenti\Arduino\libraries\PZEM004Tv30-1.1.2\src/PZEM004Tv30.h:63:7:
note: candidate: constexpr PZEM004Tv30::PZEM004Tv30(const PZEM004Tv30&)
>
>  class PZEM004Tv30
>
>        ^~~~~~~~~~~
>
>
C:\Users\maria_dchxf68\OneDrive\Documenti\Arduino\libraries\PZEM004Tv30-1.1.2\src/PZEM004Tv30.h:63:7:
note:   no known conversion for argument 1 from 'SoftwareSerial' to 'const
PZEM004Tv30&'
>
> exit status 1
>
> no matching function for call to
'PZEM004Tv30::PZEM004Tv30(SoftwareSerial&)'
>
>
>
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

sergiocntr, 2021-07-25:
Hi.
Try the basic example :

`SoftwareSerial pzemSWSerial(PZEM_RX_PIN, PZEM_TX_PIN);
PZEM004Tv30 pzem(pzemSWSerial);

void setup() {
    /* Debugging serial */
    Serial.begin(115200);
}`

diegomariani, 2021-07-25:
Don't work. I'm error here:

sergiocntr, 2021-07-25:
a) this line inside setup() is not needed and broke your sketch:
pzem = PZEM004Tv30(pzemSWSerial);

b) your declaration miss the argument:
PZEM004Tv30 pzem;

This is the good one:
PZEM004Tv30 pzem(pzemSWSerial);

Fix this two issues in your sketch.

diegomariani, 2021-07-25:
I'm try all....

don't work.

tanck's for your time.

mandulaj, 2021-07-25:
I think what might be the problem is the way we conditionally enable SoftwareSerial in the library only on some platforms (AVR and ESP8266).

See here:
https://github.com/mandulaj/PZEM-004T-v30/blob/b27887e2118f67bda1bb100e6c82a859dafda1c9/src/PZEM004Tv30.h#L49-L52

So perhaps try force enabling the Software serial by defining `PZEM004_SOFTSERIAL` manually.

```c++

#define PZEM004_SOFTSERIAL

#include <PZEM004Tv30.h>
#include <SoftwareSerial.h>

SoftwareSerial pzemSWSerial(6, 7);
PZEM004Tv30 pzem(pzemSWSerial);

void setup() {

    SerialUSB.begin(115200);

    while (!SerialUSB) { 
         delay(100); 
    }

    SerialUSB.println("start");

}

void loop() {
}
```

diegomariani, 2021-07-25:
I'm try this too....

But I muss buy a new board, while seeeduino work with 3.3V, e I muss have
5V.

Tank's for all

Diego

JFRosado, 2024-11-21:
I've tried every options suggested here and no luck :(