mandulaj/PZEM-004T-v30 #73: Problem Compiling ''PZEMHardSerial''

OctubreRojo, 2021-07-28:
I have a problem compiling PZEMHardSerial on Arduino IDE 1.8.15 for ESP32. It shows me this message: '' exit status 1
no matching function for call to 'PZEM004Tv30::PZEM004Tv30(HardwareSerial&, int, int)'  ''
Does anybody knows how can I solve this? Or which arduino IDE version should I use?

[image]

mandulaj, 2021-07-29:
Have you by chance installed the old version of the library earlier? It looks like there is some kind of a conflict and Arduino can't find the current version source files. I've just tried it on my Arduino 1.8.15 and the PZEMHardSerial example compiled OK. Also could you post a bit more of your source code.

OctubreRojo, 2021-07-29:
I think you're right. Probably what I did is that I installed olehs' library and then yours. (Not sure if I had the previous version of your library). I've just installed Arduino and your library in another pc and compiled with no problem. 
Thank you very much.

mandulaj, 2021-07-30:
No problem. I've just recently added the library to the Arduino Library Manager and I am expecting similar issues to come up. I think many people had it manually installed before and I have no idea how the Arduino IDE handles this. From my personal experience, it usually creates a new folder and then all sorts of duplicate header issues can come up. I will close your issue for now if that is ok.

closed 2021-07-30 by mandulaj