mandulaj/PZEM-004T-v30 #74: arduino mkr wifi 1010 errors in library

97kingsman, 2021-08-09:
//  _    _         _               ___     ___ 
//  \\  //         ||       /\    ||  \\  //  \\
//   \\//    __    ||      /__\   ||__//  \\__  
//    //\\  //\\   ||     //  \\  ||  \\  _   \\
//   //  \\//  \\  ||||| //    \\ ||__//  \\__// 

// A YM Labs Production

//////////////////////////////////////////////
// For Stepper Motor -  Pins                //
   const int stepPin = 3;                   //
   const int dirPin = 4;                    //      
////////////////////////////////////////////// 
// For LCD                                  //
//  Libraries                               //  
    #include <Wire.h>                       //
    #include <LCD.h>                        //
    #include <LiquidCrystal_I2C.h>          //
//  I2C Object,                             //
LiquidCrystal_I2C lcd(0x27,20,4);  // set the LCD address to 0x27 for a 16 chars and 2 line display
//////////////////////////////////////////////
//  For Ultrasonic Sensor                   //
// Trigger pin and echo pin                 //
    const int pingPin = 7;                  //
    const int echoPin = 6;                  //
// Distance                                 //
    long duration, cm;                      //
//////////////////////////////////////////////
// For Limit switch                         //
// Setting pin and value                    //        
    const int b_ls = A3;                    //
    int bls_val = LOW;                      //   
//////////////////////////////////////////////
//  For IR Sensors                          //
//  Setting the pins and values             //
    const int ir1 = A4;                     //
    const int ir2 = A5;                     //
    int val1 = 0;                           //
    int val2 = 0;                           //
//////////////////////////////////////////////
//  For timers                              // 
    long int time1, time2;                  //
    int timer = 1000;                       //
    long int dis_time = 12500*60;           //
//////////////////////////////////////////////
//  For UV light                            //
    const int uv_pin = 1;                   //
//////////////////////////////////////////////
// For Servo                                //
   const int servopin = 9;                  //
//////////////////////////////////////////////

void setup() 
{

//////////////////////////////////////////////
//Setting up the Traverser                  //
//Setting Stepper Motor pins                //
  pinMode (stepPin,OUTPUT);                 //
  pinMode (dirPin,OUTPUT);                  //    
                                            //
//Enabling pulldown for Limit Switch        //
  pinMode (b_ls, INPUT);           //
                                            //
//Set direction to backward                 //
  digitalWrite (dirPin,LOW);                //
//------------------------------------------//
//Bring the traverser to the position       //
  while (bls_val == LOW)                    //
  {                                         //
[... 494 lines cut ...]
C:\Users\hp\Documents\Arduino\libraries\lcd/LCD.h:97:26: error: lvalue required as left operand of assignment

    PORTB=((str[k])&0xF0)|0x01;

                          ^~~~

C:\Users\hp\Documents\Arduino\libraries\lcd/LCD.h:99:31: error: lvalue required as left operand of assignment

    PORTB=((str[k]&(0x0F))*16)|0x01;

                               ^~~~

Multiple libraries were found for "LCD.h"

 Used: C:\Users\hp\Documents\Arduino\libraries\lcd

 Not used: C:\Users\hp\Documents\Arduino\libraries\1027037-f5231e9d1ae62b25816934e55cc64ec8d104c803

exit status 1

Error compiling for board Arduino MKR WiFi 1010.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mandulaj, 2021-08-10:
Hi, I am sorry. How is this related to the PZEM library? The errors seem to be related to the LCD library.

97kingsman, 2023-06-14:
> 

yes i resolved it.

97kingsman, 2023-06-14:
> Hi, I am sorry. How is this related to the PZEM library? The errors seem to be related to the LCD library.

yes you r right i resolved it. i checked your reply later. thankyou for responding