mandulaj/PZEM-004T-v30 #75: Reset comunication

DavidAntonin, 2021-09-09:
Hello,
can you implement reset comunication over serial port? Once every few days the communication with PZEM freezes and I read nothing but zeros. After restarting ESP8266 everything is fine. I use HW serial.

mandulaj, 2021-09-21:
You might actually have discovered quite a serious bug! I think the problem might be with the updateValues function which rate limits the number of actual update calls. 

There is basically a piece of code that makes sure we only update once ever 200ms. And I happen to be doing it in the completely wrong way! Basically when the timer rolls over after about 50 days, the check can become invalid. It's a real shame I haven't caught it earlier. Thanks, I am going to push a fix soon

DavidAntonin, 2021-09-23:
I don't think communication ends in 50 days, I would say it's less, sometimes just a few days

mandulaj, 2021-09-24:
That is strange then... Could you verify if the values you get are actually `0` or `Nan`. Nans represent an error in the communication. But if the library returns a 0, that's the value that was reported by the PZEM chip (CRC checked and all) In such case the problem could be with the PZEM chip itself. Do you have a second PZEM module to test this with?

DavidAntonin, 2021-10-10:
I have 2 PZEM 3.0 on the same UART port. If communication fail, I get Nan from both of them.

christianBricaud, 2021-11-03:
If there is a parasite on the reception line, during the next request, there are 26 characters in the reception buffer and since only the first 25 bytes are read, the frame is incorrect and there is always a character in the buffer. All subsequent frames will be incorrect. Simple solution: empty the buffer before making a request:
In the sendCmd8 function, just before the line "_serial-> write (sendBuffer, 8); // send frame", insert this line: "while (_serial-> available ()) _ serial-> read ();"

hdimon, 2021-12-14:
Hi @mandulaj!
First, thank you very much for your library and your efforts maintaining that. 
Second, thanks for this particular fix:). I've been living with that about 1,5 years and didn't have time to investigate it myself (and I even was not sure where is error - in library or something else) so I just manually reset Mega2560 every 1,5 months.
Today I've updated one of my controllers with fixed library and if it's ok during today then likely tomorrow I'll also update the second one.
I also manually added line as @christianBricaud suggested: "while (_serial-> available ()) _ serial-> read ();".

But I want to mention that previos month I also faced case when one microcontroller stopped getting data from PZEM not after ~50 days but after 24 days. So my question is **if I can do something else in additional to your and christianBricaud fixes in order to increase reliability**? For me it's quite important, my other microcontrollers have been working more than year without resetting so I try to achieve reliability when microcontrollers are reset only due to power off rather than issues with sensors:).

iwas108, 2023-04-18:
Any solution to this problem? Me too have a connection lost after a few minutes.

mandulaj, 2023-04-20:
Hey @iwas108, could you give me more details on your problem? What error do you get after a few minutes? It could be a result of any number of issues. Does the program simply stop? Or do you start receiving garbage data, NaNs?

mandulaj, 2023-05-05:
yes this is strange indeed. during the readout, can you see both RX and TX LEDs blinking on all of the boards?

iwas108, 2023-05-10:
I fixed the problem by moving the PZEM instantiation to the local function. May be there is an issue with my code, because I was using an async libs with global object/variable.

mandulaj, 2023-05-10:
Ok, weird. could you perhaps share the full sketch you were using before?

iwas108, 2023-05-12:
This is my working code, copied from here https://github.com/Narin-Laboratory/UDAWA-Gadadar/blob/rtos%2Basync/Firmware/gadadar_esp32/src/main.cpp

The solution is, i just moved the instantiation code:
```
 HardwareSerial PZEMSerial(1);
 PZEM004Tv30 PZEM(PZEMSerial, S1_RX, S1_TX);
```
from Global Vars to local Task function. Before this solution, my code does not use any RTOS Task, just used normal void loop() and millis(), and the instantiation is in the global area above void setup(). The problem is after running around 40 minutes, the serial comm to the PZEM was stopped working, and working again after manual ESP32 restart.

```
/**
 * UDAWA - Universal Digital Agriculture Watering Assistant
 * Firmware for Actuator 4Ch UDAWA Board (Gadadar)
 * Licensed under aGPLv3
 * Researched and developed by PRITA Research Group & Narin Laboratory
 * prita.undiknas.ac.id | narin.co.id
**/
#include "main.h"
#define S1_TX 32
#define S1_RX 4

using namespace libudawa;
Settings mySettings;
Adafruit_BME280 bme;

BaseType_t xReturnedWsSendTelemetry;
BaseType_t xReturnedWsSendSensors;
BaseType_t xReturnedRecPowerUsage;
BaseType_t xReturnedRecWeatherData;
BaseType_t xReturnedPublishSwitch;
BaseType_t xReturnedRelayControl;

TaskHandle_t xHandleWsSendTelemetry = NULL;
TaskHandle_t xHandleWsSendSensors = NULL;
TaskHandle_t xHandleRecPowerUsage = NULL;
TaskHandle_t xHandleRecWeatherData = NULL;
TaskHandle_t xHandlePublishSwitch = NULL;
TaskHandle_t xHandleRelayControl = NULL;

void setup()
{
  long startMillis = millis();

  processSharedAttributeUpdateCb = &attUpdateCb;
  onTbDisconnectedCb = &onTbDisconnected;
  onTbConnectedCb = &onTbConnected;
  processSetPanicCb = &setPanic;
  processGenericClientRPCCb = &genericClientRPC;
  emitAlarmCb = &onAlarm;
  onSyncClientAttrCb = &onSyncClientAttr;
  onSaveSettings = &saveSettings;
  #ifdef USE_WEB_IFACE
  wsEventCb = &onWsEvent;
  #endif
  startup();
  loadSettings();
  syncConfigCoMCU();
  if(String(config.model) == String("Generic")){
    strlcpy(config.model, "Gadadar", sizeof(config.model));
  }
  stateReset(0);

  tb.setBufferSize(1024);

  #ifdef USE_WEB_IFACE
  xQueuePZEMMessage = xQueueCreate( 1, sizeof( struct PZEMMessage ) );
  xQueueBME280Message = xQueueCreate( 1, sizeof( struct BME280Message ) );
  #endif

  mySettings.flag_bme280 = bme.begin(0x76);
[... 1130 lines cut ...]
          serializeJson(doc, buffer);
          ws.broadcastTXT(buffer);
          doc.clear();
        }
      } 

      if( xQueueBME280Message != NULL ){
        BME280Message BME280Msg;
        if( xQueueReceive( xQueueBME280Message,  &( BME280Msg ), ( TickType_t ) 1000 ) == pdPASS )
        {
          JsonObject bme280 = doc.createNestedObject("bme280");
          bme280[PSTR("celc")] = round2(BME280Msg.celc);
          bme280[PSTR("rh")] = round2(BME280Msg.rh);
          bme280[PSTR("hpa")] = round2(BME280Msg.hpa);
          bme280[PSTR("alt")] = round2(BME280Msg.alt);
          serializeJson(doc, buffer);
          ws.broadcastTXT(buffer);
        }
      } 
    }
    vTaskDelay((const TickType_t) 1000 / portTICK_PERIOD_MS);
  }
}
#endif
```

thelightings, 2023-08-06:
> If there is a parasite on the reception line, during the next request, there are 26 characters in the reception buffer and since only the first 25 bytes are read, the frame is incorrect and there is [...]

Thank you alot. It is the problem.

thelightings, 2023-08-06:
@christianBricaud Thank you alot. It is the problem.

Embedded-Tosh, 2025-04-19:
Hi @mandulaj,
I'm experiencing the same issue, and it happens randomly — I can't figure out what triggers it. Here are my logs.

T: 2025-04-19 12:19:00, V: 237.40, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:02, V: 237.40, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:04, V: 237.40, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:06, V: 237.60, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:08, V: 237.60, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:10, V: 237.60, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:12, V: 237.70, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:14, V: 237.50, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:16, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[226796][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:19, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[229189][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:21, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[231583][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:24, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[233972][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:26, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[236361][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:29, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[239186][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:31, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[241554][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:34, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[243933][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:36, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[246322][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
T: 2025-04-19 12:19:38, V: nan, C: nan, W: nan, kWh: nan, PF: nan, Hz: nan
[248712][E][main.cpp:517] readEnergyDataTask(): [PZEM] PZEM read error
[250712][E][main.cpp:472] resetPZEMSerial(): [Serial2] Flush and Reset PZEM Serial2
E (250680) uart: uart_get_buffered_data_len(1380): uart driver error
[250815][E][main.cpp:477] resetPZEMSerial(): [Serial2] Reset Done
T: 2025-04-19 12:19:41, V: 236.70, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:43, V: 236.80, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:45, V: 236.80, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00
T: 2025-04-19 12:19:48, V: 236.60, C: 0.00, W: 0.00, kWh: 1526.29, PF: 0.00, Hz: 50.00

However, I did find a workaround: after counting to 10, I simply reset the hardware Serial2 communication, and that seems to restore the serial connection.

void resetPZEMSerial()
{
  ESP_LOGE("Serial2", "Flush and Reset PZEM Serial2");
  Serial2.flush();                                           // Flush remaining data
  Serial2.end();                                             // End the serial
  vTaskDelay(pdMS_TO_TICKS(100));                            // Small delay for cleanup
  Serial2.begin(9600, SERIAL_8N1, PZEM_RX_PIN, PZEM_TX_PIN); // Reinitialize (baud, config, RX, TX)
  ESP_LOGE("Serial2", "Reset Done");
}