mandulaj/PZEM-004T-v30 #76: Help: Address Problem

shiftux, 2021-09-24:
Hi
First off, thanks for writing this library! - Makes life so much easier!

I'm struggling with the very first interfacing of the [PZEM-004T-100A(V3.0)](https://www.aliexpress.com/item/4000330631886.html) via my [ESP32 devkit V1](https://www.aliexpress.com/item/4000281177613.html). 
I was using the (unmodified) example code in `PZEM-004T-v30/examples/PZEMHardSerial/PZEMHardSerial.ino` to test readouts of the device. However, all I am getting is:
```
Custom Address:0
Error reading voltage
```
As a fix I was trying to use `PZEM-004T-v30/examples/PZEMChangeAddress/PZEMChangeAddress.ino`, but this seems to fail as well:
```
Previous address:   0x0
Setting address to: 0x55
Error setting address.
```

I will note that at first, I was trying to use the [older library](https://github.com/olehs/PZEM004T). I'm hoping this didn't somehow mess up the addressing of the device. However, I tried it with another PZEM (on which I had not used the old lib) and the behavior is the same. This makes me think I am doing something wrong, but I can't figure out what it is...

I've attached a pic detailing my wiring setup. And my `platformio.ini` file looks like below.

Help is much appreciated! Thanks!

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
monitor_filters = direct, esp32_exception_decoder
lib_deps = 
    knolleary/PubSubClient@^2.8
    /Users/shiftux/Documents/Arduino/libraries/PZEM004Tv30/src/
    EspSoftwareSerial
```

[image]

mandulaj, 2021-09-24:
Hi, it seem that the problem is most likely in your wiring. From the picture its slightly unclear where you have connected the all the wires. Make sure that the 5V (Yellow wire) is connected to the 5V pin, Ground (Black) to a ground pin and the RX/TX (White and Red) to the TX/RX pin of the ESP32. By default those are the GPIO17 and GPIO16 respectively. [image]. Also try swapping the RX and TX around, just in case.

You should see the RX and TX LEDs flashing on the PZEM module. Also make sure the module is hooked up to the AC. The chip is actually powered by the AC, not the 5V!

shiftux, 2021-09-24:
ah wow! - the AC current has to be connected...! - did not get that! 
i feel silly! 🙈
Thanks for the quick reply!

closed 2021-09-24 by shiftux

shiftux, 2021-09-24:
the strange thing about it is that you actually see one of the 2 lights flashing, even without the AC. so I never really thought it was not powered...

mandulaj, 2021-09-26:
Yes, I agree it is slightly confusing. The board is split up into a high voltage (AC 240V) and low voltage (5V) section. These sections are powered independently but are bridged by the white Optocouplers so no electrical power is flowing between them (this is good for safety). 

The PZEM chip and the TX optocoupler is powerd by the AC section while the RX optocoupler and both the RX and TX indication LEDs are powered by the 5V line so it is expected that only RX line flashes when the AC isn't connected because that is your ESP trying to send data to the PZEM chip but not getting any reply.

Yarce008, 2022-07-31:
Hello, I have had the same problems as you, I have used the same code ... you already have the solution

Yarce008, 2022-07-31:
> Hi First off, thanks for writing this library! - Makes life so much easier!
>
> I'm struggling with the very first interfacing of the [...]

Hello, I have had the same problems as you, I have used the same code ... you already have the solution

mandulaj, 2022-08-01:
Hey @Yarce008  Could you be more specific in describing your issue? What is the exact problem? Have you tried any of the suggested solutions (swapping RX/TX, connecting the module to AC, etc.)  I can't really help you otherwise.

Yarce008, 2022-08-02:
> Hey @Yarce008 Could you be more specific in describing your issue? What is the exact problem? Have you tried any of the suggested solutions (swapping RX/TX, connecting the module to AC, etc.) I can't really help you otherwise.

hello, cordial greeting, thanks for answering...see how I'm Spanish-speaking, I have to use a translator...but I'll tell you:
I have used the different codes to make the electrical measurements. I am using a PZEM -004T-100A(V3.0) AND AN ESP32 DEVI VERCION 4
It does not make me measurements and it gives me the same error as the publication
I would like you to give me a hand .... I will prepare a video for you or upload some detailed photos I am still new to this