mandulaj/PZEM-004T-v30 #82: PZEM-004T on big electronics device

Albar0o0, 2022-02-10:
asking for my project, does PZEM-004T can be used to measure big electronic devices? for example, on a refrigerator, fan, or TV screen..? if it can, please tell me how to do it, or maybe give me a link to any papers or youtube videos, or anything that might help.

Thanks!

mandulaj, 2022-02-18:
Hey, Yes the PZEM-004T can measure large devices. In fact, that is exactly what it is designed to do. However, I am not sure that it is the best device for you to use. You will have to wire 230V AC power which, if you don't know what you are doing, can be **extremely dangerous**! Here is a basic video I found on Youtube https://www.youtube.com/watch?v=3kayhWvy6GA 

But I am gong to repeat it again: Use the PZEMs at your own risk only if you have at least a basic understanding of electricity. The module does **not** usually come with an insulating box and exposes lethal voltage which will kill you if you touch it where you shouldn't.

closed 2022-02-18 by mandulaj

mandulaj, 2022-02-18:
Have you watched the video? II don't have a link to a tutorial specifically for the PZEM module. But I am sure the internet is full of useful resources. Try learning something about electricity, then look for how to splice a current probe into the power cable of your appliance. The keywords that will probably help you are:  "safety around AC", "current transformers" But using the PZEM device with larger electronic equipment is not much different from using it for something like a small lightbulb. Just make sure to use sufficiently thick conductors.

I am afraid you will have to do some of your own research.

For a start, take a look at this wiring diagram: 
[image]

You just replace the load (the lightbulb) with your fridge or TV power cable. How to do that should not be difficult for you to figure out.

mandulaj, 2022-02-18:
If you buy the version with plastic cover (like the one in the video) it should be relatively safe. If you know the basics, only connect wires when it's not powered and don't do anything stupid you should be ok. But I always say this warning explicitly because unfortunately 220V can be deadly of not handled with respect. GreatScott has some useful videos about electricity basics https://www.youtube.com/c/greatscottlab so would start there.