mandulaj/PZEM-004T-v30 #85: ESP32 Hardware Serial doesn't work

Ncent-EP, 2022-04-03:
I am using the hardware serial on ESP32 but it seems doesn't work, the PZEM cannot give a response to ESP32. The serial monitor shows only.

`Custom Address:0
Error reading voltage
`

I was think the command that give from ESP32 isn't send properly to PZEM. please fix it, thanks :)

sergiocntr, 2022-04-03:
Hi,
There are many Esp32 working with this lib ,so check deeply your code and connections.
If everything looks fine, take a look at closed  issue #64 .

lequangthinhhaui, 2022-04-11:
Please plug the voltage, it will work when voltage >0

MegoVaTx, 2023-12-30:
Same issue Rx TX leds on sensor stay lit all the time. begin(9600,SERIAL_8N1,16,17);
does work.
Also SoftwareSerial on uno works with sensor.
Looks like Library cant communicate with sensor at all.
Tried multiple ESP32, Rx connected to TX, TX to RX, wired properly.

Code Im using
`/*
Copyright (c) 2021 Jakub Mandula

Example of using one PZEM module with Hardware Serial interface.
================================================================

If desired, a HardwareSerial handle can be passed to the constructor
which will then be used for the communication with the module.

Note that ESP32 HardwareSerial must also be provided with the RX and TX
pins.

*/

#include <PZEM004Tv30.h>
//RX 16 to white
//TX 17 to green
PZEM004Tv30 pzem(Serial2, 16, 17);

void setup() {
    // Debugging Serial port
    Serial.begin(115200);

    // Uncomment in order to reset the internal energy counter
    // pzem.resetEnergy()
}

void loop() {
    // Print the custom address of the PZEM
    Serial.print("Custom Address:");
    Serial.println(pzem.readAddress(), HEX);

    // Read the data from the sensor
    float voltage = pzem.voltage();
    float current = pzem.current();
    float power = pzem.power();
    float energy = pzem.energy();
    float frequency = pzem.frequency();
    float pf = pzem.pf();

    // Check if the data is valid
    if(isnan(voltage)){
        Serial.println("Error reading voltage");
    } else if (isnan(current)) {
        Serial.println("Error reading current");
    } else if (isnan(power)) {
        Serial.println("Error reading power");
    } else if (isnan(energy)) {
        Serial.println("Error reading energy");
    } else if (isnan(frequency)) {
        Serial.println("Error reading frequency");
    } else if (isnan(pf)) {
        Serial.println("Error reading power factor");
    } else {

        // Print the values to the Serial console
        Serial.print("Voltage: ");      Serial.print(voltage);      Serial.println("V");
        Serial.print("Current: ");      Serial.print(current);      Serial.println("A");
        Serial.print("Power: ");        Serial.print(power);        Serial.println("W");
        Serial.print("Energy: ");       Serial.print(energy,3);     Serial.println("kWh");
        Serial.print("Frequency: ");    Serial.print(frequency, 1); Serial.println("Hz");
        Serial.print("PF: ");           Serial.println(pf);

    }

    Serial.println();
    delay(2000);
}
`

MegoVaTx, 2023-12-30:
The module is powered by AC.
As mentioned earlier, Sensor is finer (tested with Swserial on Uno)
Esp32 is good tested with MeterSerial.begin(9600,SERIAL_8N1,16,17);

MegoVaTx, 2023-12-30:
I managed to get it working with:https://github.com/jaggil/Control-of-meter-PZEM-004T-with-ESP32-Modbus?tab=readme-ov-file
with ModBus lib eModbus downgraded to 1.5.0