mandulaj/PZEM-004T-v30 #89: Putting 3 PZEM modules in parallel does not work

sim0njo, 2022-06-25:
Hi,

I have 3 PZEM modules configured at address 1, 2 and 3.

I have removed the pullup resistor (R4) and also the 2 LED's from each PZEM according to this schematic (https://community.home-assistant.io/t/how-to-use-pzem004t-energy-monitor-with-esphome/107083/122).

My ESP32 hw has a 1K resistor as pullup for the optocoupler outputs and they are wired in parallel all 3. The optocoupler LEDs of the  PZEM's are driven with a MMBT2907 transistor and I use 3V3 as power for the optocoupler circuitry.

On the ESP32 I use HW serial port 2 and get following results for combinations: 1+2=ok, 1+3=ok, 2+3=ok... but 1+2+3=nok.

Anybody an idea why 3 PZEM's in parallel don't work? They are all powered by mains, I see their power light on.

thx, Jo

sim0njo, 2022-06-26:
The issue was with the LED's of the optocoupler, they work fine with 3V3 when 1 or 2 PZEM's are present, but a third causes the weakest to fail.

Solution is to apply 5V for the LED's.

closed 2022-06-26 by sim0njo

vortigont, 2022-06-27:
you can also lower R8's value to make it 3.3v capable. See [here](https://github.com/vortigont/pzem-edl/blob/main/docs/hwmod.md)