mandulaj/PZEM-004T-v30 #90: Need Code Example Read PZEM-004T V.30 from PLC Keyence KV7500+KVL21V

MFayH, 2022-07-11:
**Is your feature request related to a problem? Please describe.**
I have ordinary use the PZEM sensor with microcontroller, but I have challenges to read the sensor with PLC Keyence, and I have stucked. Any one help me?

**Describe the solution you'd like**
I have 15 PZEM to read with ModBUS RTU Communication Protocol. I have done to change the slave address through ESP32, and I have try to read from PLC, but still can't

Thank you in advance

mandulaj, 2022-07-14:
I am sorry, unfortunately, the library doesn't support the PLC platform at the moment and I am not sure it ever will. Have you considered using the ESP32 to read the values from the PZEM modules and forwarding them to the PLC? I don't know that communication protocols the PLC supports, but I am sure you can set up a UART/SPI/I2C/Wifi connection that will allow you to relay the data.

Nismonx, 2022-07-14:
Most PLCs from that generation will handle RS232 / RS485 but you'll have to understand the protocol and be able to parse data to whatever protocol you planning on using . If you have an HMI with WiFi capability then you may be able to send MQTT packages to it but this will depend on the manufacturer (Weintek). 
It's an interesting challenge you are trying to achieve so, keep us posted if you find a way around it. 

Good luck

MFayH, 2022-07-15:
> I am sorry, unfortunately, the library doesn't support the PLC platform at the moment and I am not sure it ever will. Have you considered using the ESP32 to read the values from the PZEM modules and [...]

Thank you mandulaj for your reply, So now I know that the sensor doesn't support with PLC. But for the connection from sensor to PLC is through some devices, not directly connected :
AC Voltage -> PZEM Sensor -> TTL to RS485 Converter -> KV-L21V (Keyence PLC's Serial Module) -> KV-7500 (Keyence PLC's CPU Unit)

MFayH, 2022-07-15:
> Most PLCs from that generation will handle RS232 / RS485 but you'll have to understand the protocol and be able to parse data to whatever protocol you planning on using . If you have an HMI with [...]

Thank you Nismonx for your reply, yes of course I have used the RS485 Communication Protocol (2 wire - half duplex) from PZEM Sensor to PLC, and I think it will work because the PZEM Sensor Datasheet mention it (Application Layer Protocol - Modbus RTU). But, unfortunately it doesn't work properly.

sergiocntr, 2022-07-15:
I don't know what your PLC know-how is, but I found this [link] (https://www.keyence.com.sg/support/user/plc/) where you can find support and assistance. Sorry if I don't help you more, 15 devices are enough for me  to open a ticket.

MFayH, 2022-07-15:
So my alternative solution is, I haven't use the PLC for read the PZEM Sensor. I used the ESP32 for read the PZEM Sensor and send it to database

sergiocntr, 2022-07-15:
> So my alternative solution is, I haven't use the PLC for read the PZEM Sensor. I used the ESP32 for read the PZEM Sensor and send it to database

Yes this Is a way but consider this as Plan B

MFayH, 2022-07-15:
> I don't know what your PLC know-how is, but I found this [link] (https://www.keyence.com.sg/support/user/plc/) where you can find support and assistance. Sorry if I don't help you more, 15 devices are enough for me to open a ticket.

Thank you Sergiocntr for your advice. I think enough for me to use and struggle read the PZEM Sensor with PLC's haha, because my deadline is near I have to done in October until the Data Processing.

MFayH, 2022-07-15:
> > So my alternative solution is, I haven't use the PLC for read the PZEM Sensor. I used the ESP32 for read the PZEM Sensor and send it to database
> 
> Yes this Is a way but consider this as Plan B ( my opinion of course)

Of course, this is my Plan B. Thank you GBU