mandulaj/PZEM-004T-v30 #92: ESP32 Serial2 + 14 PZEM-004T Sensor Module

MFayH, 2022-08-09:
Hi everyone. I try to read 14 PZEM-004T Sensor Module with single serial port (Serial2, RX16, TX17). But it like there is a limit for connections. just can communicate 5 sensors module randomly from 14 sensors. So if I connect the 6th sensor, the respon in serial monitor is "Sensor Error Measuring". There is any limitation of connection in Serial port? Any solution for it? Thank you before

There is my program (long, but similar, just different variables and sensor address)

`
#include <PZEM004Tv30.h>
#include <WiFi.h>
#include <HTTPClient.h>

HTTPClient http;

#define PZEM_RX_PIN 16
#define PZEM_TX_PIN 17
#define PZEM_SERIAL Serial2
#define NUM_PZEMS 13

PZEM004Tv30 pzems[NUM_PZEMS];

const char WIFI_SSID[] = "OPPO Reno6 Pro 5G";
const char WIFI_PASSWORD[] = "34567890";

void setup() {
  Serial.begin(9600);
//
  // For each PZEM, initialize it
  for(int i = 0; i < NUM_PZEMS; i++) {
        pzems[i] = PZEM004Tv30(PZEM_SERIAL, PZEM_RX_PIN, PZEM_TX_PIN, 0x01 + i);
  }
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected WiFi with IP Address: ");
  Serial.println(WiFi.localIP());
}

float calib = 3.5;

float vGVDMain, cGVDMain, pGVDMain, eGVDMain, fGVDMain, pfGVDMain;
float vGVDClean, cGVDClean, pGVDClean, eGVDClean, fGVDClean, pfGVDClean;
float vRVD901, vSVD901, vTVD901, cRVD901, cSVD901, cTVD901, pRVD901, pSVD901, pTVD901, eRVD901, eSVD901, eTVD901, fRVD901, fSVD901, fTVD901, pfRVD901, pfSVD901, pfTVD901;
float vRVD301, vSVD301, vTVD301, cRVD301, cSVD301, cTVD301, pRVD301, pSVD301, pTVD301, eRVD301, eSVD301, eTVD301, fRVD301, fSVD301, fTVD301, pfRVD301, pfSVD301, pfTVD301;
float vRPRC, vSPRC, vTPRC, cRPRC, cSPRC, cTPRC, pRPRC, pSPRC, pTPRC, eRPRC, eSPRC, eTPRC, fRPRC, fSPRC, fTPRC, pfRPRC, pfSPRC, pfTPRC;
float vRPMB, vSPMB, vTPMB, cRPMB, cSPMB, cTPMB, pRPMB, pSPMB, pTPMB, eRPMB, eSPMB, eTPMB, fRPMB, fSPMB, fTPMB, pfRPMB, pfSPMB, pfTPMB;

void VD901() {
  Serial.print("VD901");
  Serial.print(" - Address : ");
  Serial.println(pzems[0].getAddress(), DEC);
  Serial.println("===================");

  vRVD901 = pzems[0].voltage()+calib; vSVD901 = pzems[1].voltage()+calib; vTVD901 = pzems[2].voltage()+calib;
  cRVD901 = pzems[0].current(); cSVD901 = pzems[1].current(); cTVD901 = pzems[2].current();
  pRVD901 = pzems[0].power(); pSVD901 = pzems[1].power(); pTVD901 = pzems[2].power();
  eRVD901 = pzems[0].energy(); eSVD901 = pzems[1].energy(); eTVD901 = pzems[2].energy();
  fRVD901 = pzems[0].frequency(); fSVD901 = pzems[1].frequency(); fTVD901 = pzems[2].frequency();
  pfRVD901 = pzems[0].pf(); pfSVD901 = pzems[1].pf(); pfTVD901 = pzems[2].pf();

  if(isnan(vRVD901)){
      Serial.println("Sensor Error Measuring");
    } else {
      Serial.print("Voltage R : "+String(vRVD901)+"V");
      Serial.print("Current R : "+String(cRVD901)+"A");
      Serial.print("Power R   : "+String(pRVD901)+"W");
      Serial.print("Energy R  : "+String(eRVD901, 3)+"kWh");
      Serial.print("Freq R    : "+String(fRVD901, 1)+"Hz");
      Serial.println("PF R    : "+String(pfRVD901));
[... 322 lines cut ...]
    }
  } else {
    Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
  }

  http.end();
}

void loop() {
  VD901();
  delay(2000);
  VD301();
  delay(2000);
  PRC003A();
  delay(2000);
  PMB001C();
  delay(2000);
  GVDMain();
//  delay(2000);
//  GVDClean();

  Serial.println();
  delay(5000);
}
`

vortigont, 2022-08-09:
Check [LINKS](https://github.com/mandulaj/PZEM-004T-v30/blob/master/LINKS.md) section for additional information. ESP's gpio pin can't drive many opto's in parallel due to current limitation. You need HW MOD something like [this](https://github.com/vortigont/pzem-edl/blob/main/docs/hwmod.md). Or if you do not like modding many pzems it could be better to use external bipolar transistors of FETs to drive serial line (i.e. [like here](https://github.com/olehs/PZEM004T/wiki/Connecting-PZEM-to-ESP8266) ) If you use esp32 than you might wanna check async [event-driven library](https://github.com/vortigont/pzem-edl) for communicating with multiple pzem devs.

MFayH, 2022-08-09:
Thank Vortigont for your comment, it's not possible for me to mod the R8 into 300-500k ohm, so the possibility is to remove the R4 yeah? So I removing 13 of 14 PZEM R4's right?

MFayH, 2022-08-09:
Or I use all 3 serial ports of ESP. So each port connect to 5 PZEMs. It's tolerated? Sorry for my messy type

vortigont, 2022-08-09:
R8 and R4 pulls different lines - rx and tx. You can't tune one and omit the other. It might be OK to pull more than 5 devices with only removing R4, but I'm not sure for how many. You can experiment.
I think it would be better (and safe) to use some FET level-shifter as a buffer. You can find lot's of those on aliexpress.

vortigont, 2022-08-09:
> Or I use all 3 serial ports of ESP. So each port connect to 5 PZEMs. It's tolerated?

possible, but it may be unstable. Too close to the edge :)

MFayH, 2022-08-10:
Is there any problem if I connect the L-N of PZEM with 2 of 3 phase wire (RST)? The actual is
1. PZEM 1 -> R-S
2. PZEM 2 -> S-T
3. PZEM 3 -> R-T

MFayH, 2022-08-10:
Because while I protoyping with 14 PZEM Sensor, the connection of L-N of PZEM is R-N (neutral), and there is no problem with serial connection

MFayH, 2022-08-10:
[image]

Red highlight happened if I connect the 6th PZEM, the program just read the last.

[image]

This is the wiring

vortigont, 2022-08-10:
@MFayH quite a nice setup :)
Sorry, but I did not get what is "RTS"?
You also have some read errors even before connecting 6th device. Try using bipolar transitor for buffering, would be much safer this way.

MFayH, 2022-08-10:
RST I mean Phase 1 phase 2 and phase 3 I'm sorry for that :).
The buffer is for just RX? Or both?

MFayH, 2022-08-10:
Can you give example circuit for the transistor? It's very helpful

vortigont, 2022-08-11:
For TX (from ESP) it's just simple emitter-follower
[image], any typical low-voltage pnp will do I suppose.

For TX (from PZEMs) it won't work since each PZEM has it's own optocoupler that should pull the line for all paralleled devices, so you have to remove R4's. Might even need to remove R9's for such a lot of devices. R9 is sinking current through TX LED.

vortigont, 2022-08-11:
@MFayH you may also find this [scheme](https://github.com/TheHWcave/Peacefair-PZEM-004T-/blob/main/PZEM004T-orig.pdf) quite handy to check on

vortigont, 2022-08-11:
> Is there any problem if I connect the L-N of PZEM with 2 of 3 phase wire (RST)? The actual is
> 
>     1. PZEM 1 -> R-S
> 
>     2. PZEM 2 -> S-T
> 
>     3. PZEM 3 -> R-T

Now I got your point. You mean feed PZEM with phase-to-phase wires. That's a good question actually. I see this issues for such a setup
 - interphase voltage is 380 V, polycap (yellow one) that works as power source for PZEM migth not be rated for this. Need to check.
 - need to check specs for the sensor chip and Vsens divider if it's capable to measure >380V
 - I really doubt how it will integrate power/energy values for phase-to-phase current. Expect it to produce some strange results.
 
Pls, keep us posted if u dare to try this setup. I'm really interested with the outcome :)

Nismonx, 2022-08-11:
> Is there any problem if I connect the L-N of PZEM with 2 of 3 phase wire (RST)? The actual is
> 
> 1. PZEM 1 -> R-S
> 2. PZEM 2 -> S-T
> 3. PZEM 3 -> R-T

The pzem is designed to work at 120vac or 240vac, Phase and Neutral. In other words:
L1 to N
L2 to N
L3 to N

In your case you are running of an RST transformer without the Neutral, effectively  a Delta system.

Even if you nominal voltage is below 240vbc for example
L1 to L2 = 240vac
Or
R to S = 240vac

I doubt you'll getting anything out of it.

But like @vortigont mentioned, it would be interesting to see what you get out of it.

Just be safe and ask an electrician if you are uncertain.

Good luck 👍

MFayH, 2022-08-12:
Yes you all @vortigont @Nismonx got my point about phase-to-phase connection that I have installed. I think isn't affected to the PZEM because the voltage is in range 80~260V AC. but I'll try to connect again with Phase-to-Neutral as my protoyping before.
So, for the solution I'll try this step.
1. Program trial-error like #40 issue.
2. Change phase-to-phase -> phase-to-neutral.
3. Separate PZEM connection by 5 per group, and to switching I'll use relay changeover.

Appreciate to you all for excited to this issue. Thanks

MFayH, 2022-08-12:
And when I try to call all of my PZEM (void) this is happen

> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
14:01:14.296 -> 
14:01:14.296 -> Core  1 register dump:
14:01:14.296 -> PC      : 0x400d2fec  PS      : 0x00060f30  A0      : 0x800d3071  A1      : 0x3ffb26d0  
14:01:14.296 -> A2      : 0x3ffc32b8  A3      : 0x3ffc3828  A4      : 0x00000000  A5      : 0x0000000a  
14:01:14.296 -> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d2fe8  A9      : 0x3ffb26b0  
14:01:14.343 -> A10     : 0x00000000  A11     : 0x3ffb26e3  A12     : 0x00000008  A13     : 0x0000d880  
14:01:14.343 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
14:01:14.343 -> EXCVADDR: 0x00000000  LBEG    : 0x400844ed  LEND    : 0x400844f5  LCOUNT  : 0x00000027  
14:01:14.343 -> 
14:01:14.343 -> 
14:01:14.343 -> Backtrace:0x400d2fe9:0x3ffb26d00x400d306e:0x3ffb2710 0x400d31e5:0x3ffb2730 0x400d27ae:0x3ffb2750 0x400d2c93:0x3ffb2800 0x400d950d:0x3ffb2820 
14:01:14.343 -> 
14:01:14.343 -> 
14:01:14.343 -> 
14:01:14.343 -> 
14:01:14.343 -> ELF file SHA256: 0000000000000000
14:01:14.343 -> 
14:01:14.343 -> Rebooting...
14:01:14.343 -> ets Jun  8 2016 00:22:57
14:01:14.390 -> 
14:01:14.390 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:01:14.390 -> configsip: 0, SPIWP:0xee
14:01:14.390 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:01:14.390 -> mode:DIO, clock div:1
14:01:14.390 -> load:0x3fff0030,len:1344
14:01:14.390 -> load:0x40078000,len:13864
14:01:14.390 -> load:0x40080400,len:3608
14:01:14.390 -> entry 0x400805f0

But, when I call just 6 PZEM (void) with 5 PZEM connected to ESP32. It's work fine. Any comments?

MFayH, 2022-08-12:
Is this because network connection code? I use <Wifi.h> and <HTTPClient.h> library to send the data to database.

vortigont, 2022-08-14:
What do you mean `PZEM (void)`? Stacktrace shows some null reference (add 0x00000), try to narrow down when it happens, in `setup()` or in `loop()`
Actually creating large number of objects on stack is not a good idea, especially if you need memory on things like `HTTClient`. Better to create objects dynamically and poll it one by one.
You may check this [pool example](https://github.com/vortigont/pzem-edl/tree/main/examples/03_MultiplePZEM004), but it is another library.

MFayH, 2022-08-15:
I mean call PZEM read function 

> What do you mean `PZEM (void)`? Stacktrace shows some null reference (add 0x00000), try to narrow down when it happens, in `setup()` or in `loop()` Actually creating large number of objects on stack [...]

MFayH, 2022-08-19:
Update of this issue. I add wiring for changeover relay for separate RX pin of ESP32 to read from sensor 5 by 5 until 13 sensors. 1-10 sensors are working properly, but 11th to 13th sensor said "Sensor Error Measuring". I have flush the serial pin and begin again, but doesn't work. Any suggest?

[image]

Additional Wiring for Relay Changeover
[image]

MFayH, 2022-08-19:
The push button symbol just for simulation. The actual R1 activated by GPIO 18 and R2 activated by GPIO 19 of ESP32

MFayH, 2022-08-19:
Thank you everyone. Problem has solved. 
Countermeasure 👍 :
1. Additional Relay Changeover 2 Channel for separate sensor TX to RX ESP Connection 5 by 5
2. Flush Serial ports in every 5 sensors measurement dan begin the serial again

Very pleasure interact with amazing guys like you. Thanks in advance

nextstationcl, 2023-08-15:
Same problem, I only receive two measurements regardless of the directions of each sensor.