mandulaj/PZEM-004T-v30 #93: Invalid values

mlaci001, 2022-08-14:
Hello
When reading the device, I get the following values:
Voltage: 231.900
Current: 0.212
Power: 0.700
Energy: 0.002
Freq: 50,000
PF: 0.010
The real consumption is 2.9 W (measured with a commercial power meter).
According to the values, 231.9(Voltage)*0.212(Current)=49.16W would come out, which is impossible.
Has anyone encountered such a problem?
Thanks.

mandulaj, 2022-08-14:
When you take the simple product of Voltage and Current, you get something called Apparent Power (`S`). If you have a resistive load, the voltage  and current are in phase and as a result the Real power (`P`) is equal to the Apparent Power (`P=S=V*I`).

However, in many cases, voltage and current are not in phase. This means that the Apparent Power contains two components, The Real power (`P`) and something called Reactive Power (`Q`). The Apparent Power `S` is complex with `theta` being the phase shift, but we often summarize the relationship using the power factor `PF=cos(theta)`: `P = |S| * cos(theta) = |S| * PF` The Reactive Power `Q` manifests as this extra 'phantom' current. However don't be fooled, the current is really there! It just does not do any useful work and instead just moves the reactive power back and forth between the load and source. 

In big power networks, this is a problem because you need to have wires sized sufficiently in order to handle both the Real and Reactive currents despite only the Real current doing useful work! So ideally your `PF` is 1 and your Apparent Power is equal to the Real power giving you maximum transmission efficiency. This is why Power Factor correction is a big business! However, you as a consumer don't have to worry about it. You will anyways only pay for the Real Power.  

[image]

You can read more about this on [Wikipedia: Power Factor](https://en.wikipedia.org/wiki/Power_factor).

If I am not mistaken, you are probably powering some kind of low-power switching device, something capacitive, or even just nothing (effectively capacitive).  This results in a horrible power factor (0.01). If you multiply it by the Apparent power you get `231*0.212*0.01 = 0.5W` which is close to the reported Real power (0.7W) and what your commercial power meter says (2.9W). 

As to why all these numbers are not exactly equal, first of all, the low PF is usually not very accurate, I would like to one day measure its true resolution, but the discrepancy between the calculated 0.5W and reported 0.7W is likely just a numerical error of the resolution (0.01 vs 0.014). And second, the PZEM module is not great at measuring low currents overall, plus it has some power consumption on its own. So you cant trust it too much for the first few watts. It does get quite accurate for larger loads though. 

So in conclusion, just use the Power: 0.7 value to get your Real Power and don't bother with reactive power.

mlaci001, 2022-08-15:
Köszönöm szépen, most már sokkal jobban értem.

vortigont, 2022-08-19:
> if I want to measure 3 phases, then the PZEM modules must be powered from the phase to be measured (230V), or they can all be powered from the same branch, regardless of the branch to be measured ?

No! You should power each device via same Line you are measuring. It is important to match current/voltage source.

> The RS485 interface must be operated from an external power supply, or +5V can somehow be obtained from the PZEM ?

Always use external supply! It's critical for safety to avoid any possible path through measured device and attached serial line.

And just to add 5 cents to Mandulaj's post for most home appliances with switching power supplies it is not related to reactive power at all since those PSU's has rectifying bridge before capacity load. Their real power is equal to apparent power but it works in pulsed mode, i.e. it draws power for only a short period of time, the rest time it's idle with a current equal to zero. So you consider PS as a ratio of pulsed load, i.e. you could achieve PF 0.5 for the ordinary incandescent lamp if you connect it to AC power via rectifying diode.

mlaci001, 2022-08-19:
All clear!
Thank you very much !

closed 2022-11-25 by mlaci001