mandulaj/PZEM-004T-v30 #94: How do I reset the energy meter values?

mosess28, 2022-08-19:
I want the energy meter to reset every time it is turned off, How do I do that?

mandulaj, 2022-08-20:
It depends on your setup. There is no way to do this automatically. I guess you could reset the module during the initialization on boot.  This assumes your MCU turns on and off simultaneously with the PZEM module. If your MCU stays on constantly, you could detect the PZEM as off when the communication fails and then next time the communication succeeds send a reset command. However, I wouldn't recommend this as there is a chance that you get false positives. Otherwise, there isn't a way to detect when the module turns on/off alone..

mandulaj, 2023-02-02:
Has your issue been resolved? If so please kindly close this issue. Thanks :)

closed 2024-03-26 by mandulaj