mandulaj/PZEM-004T-v30 #95: Max value of current from the 100A Coil Transform output at 100A circuit current.

sheminasalam, 2022-09-12:
Hi there, What will the current output of the closed type 0-100A type coil transformer if the current in the circuit is the max at 100A?  I know the theory behind the coil transformer used with PZEM module to find the value of current in the circuit. But I don't know the number of turns in the closed type coil transformer given along with pzem?

To avoid confussions, the following is the pic of the pzem module I have.
[image]

morganflint, 2022-10-09:
The CT ratio is 1:1000 (100A primary => 100mA secondary). You can see it in the label of the split current transformer (for example [here](https://innovatorsguru.com/pzem-004t-v3/)), as the rest of the hardware is the same for both CT versions.

mandulaj, 2023-02-02:
As your question been answerd? If yes, you can close the issue.