mandulaj/PZEM-004T-v30 #98: Wrong value

ckvsoft, 2022-11-26:
The values ​​are displayed incorrectly. I have tried several PZEM always the same result.
If I consume 61 watts according to the query, I get 0.54 amps. The question now is whether there is something wrong here or whether the PZEM values ​​are determined incorrectly.

lg Chris

vortigont, 2022-11-26:
This should be placed in FAQ :)
It is expected with any non pure resistive loads. PZEM provides Mean Square values for such types of loads along with PF (power factor) value. Try your tests with incandescent lamp or kettle.

ckvsoft, 2022-11-26:
Thx but
what else can i see when measuring with an incandescent lamp or a kettle?
whether the displayed values ​​are 100% correct is another issue.
only if the values ​​obviously do not match is something wrong.
At 94 watts (various standby devices) it shows me 0.54A. Should be closer to 0.4

At 1188 watts when the microwave is on, it is 5.61A should be 5.17

At 4314 watts it shows me 19.66A. Should be 18.76.

vortigont, 2022-11-27:
Check PF value, if it is less than 1, than a simple P=I*U equation is not applicable here. Current is not constant value in time for such devices.
Kettle is almost constant linear load, so should give more expected current values.

ckvsoft, 2022-11-27:
OK thanks
Everything seems a little clearer.
Thus, the approximately correct value would be determined with I*PF, right?
example
94 watts shows 0.54A. PF = 0.74
So 94 * 0.74 = 0.399 = ~0.4
Or is that just a coincidence? ;-)

That would really be a point for the FAQ

LG Chris

vortigont, 2022-11-27:
Yeah, it works fine for rough approximation. Although real situation might be different depending on load type.
BTW you should not consider it as real current is I*PF. It's more like _the load is drawing `I` amps for 3/4 time duration of AC period if `PF` is 0.75_. Or even more complex for inductive throttled loads like electric drives with power control. `U,I,P,` values that PZEM returns is an [RMS](https://en.wikipedia.org/wiki/Root_mean_square) values over time period.

mandulaj, 2023-02-02:
Added a comment about this in the FAQ, closing now. Thanks @vortigont

closed 2023-02-02 by mandulaj