mandulaj/PZEM-004T-v30 #99: Only read voltage

fadilhim, 2022-12-03:
**Describe the bug**
Hi, im new to arduino, here i want to measure power and energy from my sensor.
The measurement only return voltage value, while i need the power and energy measurement.
im wiring my D3 pin to Tx of pzem and D4 pin to Rx of pzem.
any clue would help! thank you

**To Reproduce**
Steps to reproduce the behavior:
`PZEM004Tv30 pzem(D3, D4, 0x01);
PZEM004Tv30 pzem2(D5, D6, 0x02);
PZEM004Tv30 pzem3(D7, D8, 0x55);

float calculate_power() {
  //main energy meter
  float voltage = pzem.voltage();
  float current = pzem.current();
  float power = pzem.power();
  float energy = pzem.energy();
  float frequency = pzem.frequency();
  float pf = pzem.pf(); 
  
  //energymeter 2
  float voltage2 = pzem2.voltage();
  float current2 = pzem2.current();
  float power2 = pzem2.power();
  float energy2 = pzem2.energy();
  float frequency2 = pzem2.frequency();
  float pf2 = pzem2.pf();
  
  // energy meter 3
  float voltage3 = pzem3.voltage();
  float current3 = pzem3.current();
  float power3 = pzem3.power();
  float energy3 = pzem3.energy();
  float frequency3 = pzem3.frequency();
  float pf3 = pzem3.pf();

  // float voltageResult = 0 + voltage + voltage2 + voltage3;
  // float currentResult = current + current2 + current3;
  float powerResult = 0;
  if (power > 0) {
    powerResult += power;    
  }
  if (power2 > 0) {
    powerResult += power2;
  }
  if (power3 > 0) {
    Serial.println("Add power 3");
    powerResult += power3;
  }
  Serial.println("");
  Serial.println("POWER: " + String(power));
  Serial.println("ENERGY: " + String(energy));
  Serial.println("VOLTAGE: " + String(voltage));
  Serial.println("PF: " + String(pf));
  // Serial.println("POWER2: " + String(power2));
  // Serial.println("POWER3: " + String(power3));  
  Serial.println("POWER RESULT: " + String(powerResult));  
  Serial.println("------------------------------------");
  Serial.println();

  return powerResult;
}`

**Expected behavior**
The power and energy value should return the measurement from pzem.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Board: Wemos D1 R2

sergiocntr, 2022-12-03:
Hi Fadil ,welcome here!
Look at [this](https://github.com/mandulaj/PZEM-004T-v30/issues/54) issue.

mandulaj, 2023-02-02:
Did you connect the Current transformer correctly? You have to put one of the power wires through the CT coil in order to measure the current and thus Power,Energy and PF...
[image]