mandulaj/PZEM-004T-v30 #100: ESP32-WROOM-32 (DOIT ESP32 DEVKIT V1) + PZEM-004T-100A(V3.0) - all returned values are NaN

AlexGnatko, 2022-12-10:
**Describe the bug**
Can not use PZEM-004T-100A(V3.0) with ESP32

**To Reproduce**
1. Copy/paste the `HardwareSerial` code snippet from README.md and create a sketch
2. Connect PZEM-004T: 5V, GND, RX G16, TX G17
3. Connect to 230V mains (checked with a voltmeter that the voltage is present)
4. Compile/run the sketch on ESP32
5. Observe the serial monitor

**Expected behavior**
Every two seconds I should see at least some output in the serial monitor, instead I see this:
```
Custom Address:0
Error reading voltage

Custom Address:0
Error reading voltage
```
Every two seconds the TX LED blinks on the PZEM-004T board. But all data returned by the PZEM004Tv30 object are NaN.

**Desktop (please complete the following information):**
 - Windows

Is there a way to debug this more deeply?

sergiocntr, 2022-12-11:
Hi Alex ,welcome here.
In the example folder there is a sketch named change address :with that  read your pzem address,maybe you set a wrong one in the constructor.

iwas108, 2023-01-12:
Any update on this? Because when I tried it with Arduino Nano on pin 11 and 12 it also returned address 0, error reading voltage.

mandulaj, 2023-01-19:
Hey @AlexGnatko sorry for the late reply but try swapping the RX/TX pins. G16 (ESP RX pin) should be connected to the PZEM TX pin and vice versa.

mandulaj, 2023-01-19:
@iwas108 First of all, the Arduino Nano, has only one hardware serial on pins D0 and D1 (generally used for sending things through the USB to your computer console) You will have to use Software serial with the pins 11 and 12. Check out the [PZEMSoftwareSerial](https://github.com/mandulaj/PZEM-004T-v30/tree/master/examples/PZEMSoftwareSerial) for more info. Once you do that, make sure to check the following:

- The PZEM is connected to 230V
- You have connected both the 5V and GND line correctly
- Your RX/TX lines are connected the right way (if not sure, try swapping them)
- You are using the correct unique address (if using multiple PZEMs on one Modbus line)

ahmedreghini, 2023-05-31:
> 

> Hey @AlexGnatko sorry for the late reply but try swapping the RX/TX pins. G16 (ESP RX pin) should be connected to the PZEM TX pin and vice versa.

for anyone had this problem this is the right solution (thanks mr @mandulaj :) )

jonathanpalomino, 2023-08-02:
pues a fecha de hoy yo tengo el mismo problema

mandulaj, 2023-08-08:
@jonathanpalomino Could you check that:
1. The PZEMs are connected to the AC voltage
2. The 5V and GND are connected correctly
3. The RX/TX are connected correctly (also trying swapping them)
4. The RX LED should blink every time the MCU tries reading flowed by the TX LED blinking with the response