mandulaj/PZEM-004T-v30 #101: Negative power, for consumption or production measurement

francfly12, 2023-02-01:
I have a PZEM004 connected to an Arduino Nano, and it works like a charm, thanks for this library.
Nevertheless I have noted that the current flow is always indicated as positive, even if I reverse the wire in the current sensor.
Of course, I know it reverse 50 times per second, but this is not the point.
For my application, I need to know which way the current flows (in or out from my installation including a solar production system)
I had that using a different technology and using Emon CMS circuits and current measurements which were either positive or negative, depending on power flow (in or out, consumption or production).
But i turned to the PZEM004 because it's more precise and I need some new features.
Do you know if it is possible to tweak the library so that current is indicate positive or negative, depending on the flow.
Or perhaps this is embedded in the PZEM software and no mod is possible ?
Thanks for comments or help.

mandulaj, 2023-02-02:
Hey @francfly12. This issue was already raised in #2 and we seem to have unfortunately hit a deadend. In order to know the direction of energy flow, you need to know the phase shift between voltage and current but the  PZEM firmware running on the chip does not expose this information. 

For some reason, it only provides the power-factor which is the cosine of the phase-shift and this effectively destroys the energy direction information. People have proposed ways of cleverly using two PZEM power meters to calculate the solar production such as here https://github.com/mandulaj/PZEM-004T-v30/issues/2#issuecomment-1022552435 However unfortuantely, you won't be able to do it directly in the PZEM. 

Lets hope PZEM 4 will come with this functionality.

mandulaj, 2023-02-02:
As there is not much that can be done about this, I will close this issue.

closed 2023-02-02 by mandulaj

francfly12, 2023-02-02:
Ok! thanks a lot for the reply and for the links !