mandulaj/PZEM-004T-v30 #102: reset energy, set max power , read alarm flag.

iw2fvo, 2023-02-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

iw2fvo, 2023-02-07:
good day to all,
I am using PZEM 004T v3.0 connected to esp 8266 loaded using Arduino IDE and relevant example sketches.
I am not so skill in using "c" and IDE so I would like if someone could indicate to me the code line to be used for :
> reset accumulated enrgy
> set MAX power
> read alarm flag
Thank a lot in advance for any answer.
Regards
iw2fo(at)yahoo.com

sergiocntr, 2023-02-08:
Ciao.

> > reset accumulated enrgy  ->  pzem.resetEnergy();
> > set MAX power    -> pzem.setPowerAlarm(watts);
> > read alarm flag   ->  bool alarm = pzem.getPowerAlarm();

Try  and let me know if you need more help.

iw2fvo, 2023-02-10:
Thank a lot for the suggestions !   It is OK.I am working now to transmit all the data using the E32 433T30D  LoRa module: do you have some sample code for me please ?Best regards,Ambrogio

sergiocntr, 2023-02-10:
Hi Ambrogio
nice to hear that it worked.
For LoRa  I have no code, I've never used that device. I've searched a bit and I can assure you that Renzo has everything you need  [https://github.com/xreef/LoRa_E32_Series_Library](https://github.com/xreef/LoRa_E32_Series_Library)

margohaux, 2023-03-13:
hi! i wanted to reset the energy but the code is error. may i know if you have a reference code to reset energy?