mandulaj/PZEM-004T-v30 #103: Read Only Current

ChEfstr, 2023-02-14:
Hello to everyone,

I was wondering if there is a hack to power the chip from something other than an 220 V AC Power Supply in order to read only current.

For my application i dont have access to AC Power and i want to use the board just for reading current.

Thanks,
Chris

Nismonx, 2023-02-14:
Can you elaborate about the application you are planning to use?
To generate current you'll need some sort of pressure  either AC or DC. PZEM-004 won't work with dc, you'll better off looking for a Shunt type ammeter.

If dealing with AC the PZEM should work with as little as 80vac but don't quote me on that as it's been a while.

ChEfstr, 2023-02-14:
Thanks for the fast response!

Yes it's a DC application. I just had the PZEM-004 lying around and i thought i could maybe make it work, plus the serial connection helps as i want to have multiple amp sensors on the same ESP32. Maybe I will incorporate some kind of inverter or just go to another type of amp meter as you suggested.

closed 2023-02-14 by ChEfstr

Nismonx, 2023-02-14:
PZEM-017 is your best bet as it is designed for DC.

sergiocntr, 2023-02-15:
or acs712 maybe?